MarlinFirmware/Marlin #14790: [BUG] Marlin 2.0 Bugfix Compile error ESP on BIGTREETECH SKR PRO

dant220, 2019-07-31:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

RudyBenner, 2019-07-31:
Your configuration.h file compiles here successfully. VSCode

dant220, 2019-07-31:
you need to include the variant.h file. I enabled the following:
#define SERIAL_UART_INSTANCE    6 //ex: 2 for Serial2 (USART2)
I should have included this in the note.

shitcreek, 2019-09-11:
Would you mind closing the issue if it has been resolved.

closed 2019-09-11 by dant220