MarlinFirmware/Marlin #14794: [BUG] Long BEEP on Startup and RESET

xenon2008, 2019-07-31:
### Description
I have a new (and my first SKR 1.3) for my Ender 3 Pro.
So i downloaded the latest Marlin 2.0 Bugfix and take alle settings like in all Youtube Videos.
Actually everything works pretty well so far..

But only one Thing... When i turn on my Ender 3 Pro a long terrible BEEP sounds.
This also happens when i press the Reset Button on the SKR 1.3.

The Speaker works proberly fine when i store Settings or other things!

I also tried to uncomment the Speaker but, the terrible BEEP is still here.

Then i have googled some time and found this:
https://github.com/MarlinFirmware/Marlin/issues/13965

Is this an Bug or have i made a mistake?

### Steps to Reproduce

1. [First Step]
Download latest Marlin 2.0 Bugfix and make alle Settings for an SKR 1.3 and copy the FW to the SD Card!
2. [Second Step]
Power on the Ender 3 Pro or Press the Reset Button
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ManuelMcLure, 2019-07-31:
This is not a bug. The problem is that the display makes a beep until it is properly initialized, and Marlin cannot initialize it until it gets control from the bootloader. So there is nothing that can be done in Marlin to fix this - it requires a hardware fix.

For a ReprapDiscount Smart Full Graphics display the modification listed in #13965 fixes the problem. Someone will have to investigate whether there is a similar fix that applies to the Ender 3 display.

xenon2008, 2019-07-31:
But before i also had Marlin 2.0 Bugfix on my STOCK Melzi Board with Stock Display and there was nothing like this!
(Sure an older Release of Marlin 2.0 Bugfix..)

ManuelMcLure, 2019-07-31:
The default Melzi board has a much faster bootloader so Marlin gets control much faster after a restart.
The Smoothieware bootloader used by the SKR board takes several seconds to check the onboard SD card before passing control to Marlin.
Also, I believe the default settings for processor pins on AVR and LPC1768 chips are different - one is active low after a reset, the other is active high.

xenon2008, 2019-07-31:
okay, thanks for the explanation
So I do not need to worry about something breaking or broken or possibly misconfigured?
did you listen to the video?

ManuelMcLure, 2019-07-31:
Nothing to worry about. It's annoying but should not affect anything other than your hearing.

xenon2008, 2019-07-31:
a Friend told me that on his Ender 3 he had disabled the Speaker and no Sound coms when he power on the Ender....

ManuelMcLure, 2019-07-31:
How did he disable the speaker?

xenon2008, 2019-07-31:
he has uncommented it with //

ManuelMcLure, 2019-07-31:
That is surprising. There is no way that that could take effect until Marlin boots which is after the bootloader completes.

Have you tried the same change? Also, on his E3 Pro did he also replace the board with an SKR 1.3?

xenon2008, 2019-07-31:
yes i already has uncommented the speaker in my FW and uploaded it again... but same issue.
And yes he also has a SKR 1.3 with TMC2208 Driver like me.

He sendt me his FW, and tommorow i will try to upload his FW to my Ender 3 Pro ...
because I simply have no other idea

ManuelMcLure, 2019-07-31:
One thing you can check is whether the first time you reboot after putting the new firmware on the SD card the beep is longer. In my experience that should be the case, since the bootloader takes longer to start Marlin when it is doing a firmware update.

xenon2008, 2019-07-31:
Okay thanks for this hint!

I will try this tommorow and give you Feedback here ;)

thinkyhead, 2019-07-31:
The best we can do is move the `BEEPER_PIN` init to the earliest point possible, and make a note on our Troubleshooting page about the issue.

closed 2019-07-31 by thinkyhead