MarlinFirmware/Marlin #14795: Quality issue Marlin 2.0 and 1.1.9 // extrusion too short // infill has gaps

mulski, 2019-07-31:
**Description**

Hi there, this is my first post here on github so please do not be harsh. 

Im having issues with my printers I built at my university here it germany. Both printer have the same quality issue. Im troubleshooting this since one year now, but couldnt find a solution. I tried it with the help of reddit, Facebook and the german 3D Druck-Community. The Cummunitys gave me tipps Like: speed, acc, turn coasting off and so on. Of course I tried everything but wasnt able to fix it. Now I am looking for help again.

Printer1: Heavily modified Hypercube by Tech2C (corexy, direct extruder) Titan Extruder E3DV6 Hotend Nozzle:0.3mm
Board: Bigtreetech SKR 1.3 with TMC2208
Marlin 2.0

Printer2: Hypercube by Tech2C (bowden extruder) E3DV6 Hotend Nozzle:0.4mm
Board: MKS GenL with TMC2208
Marlin 1.8.9

Slicer: Cura 4.1.0 (Simplify3D, Slic3r)

You can clearly see the artifacts in every corner. Another issue is the Infill. This is for sure related to the same problem.
I think the extruder stops a bit too early because it is at the end of the linear movement. Not enough material is pushed out leaving gaps behind.
The top of the cube looks perfect.

What I did:
calibrating the extruder
diffrent speed, acceleration, jerk (slower and faster)
different slicer (Simplify3D, Slic3r)
different Filament (different PLA and PETG)
different board: MKS Sbase 1.3, MKS Base 1.4, MKS GenL, SKR 1.3
different steppermotor driver (TMC2208, A4982, LV8729, DRV8825)
configurate comletly new Marlin
tighten belts and screws
changed the extruder to relative and absolut modus
tried different coolingsystems (more fans, different ducts)
tried different Startcodes
different slicer settings
enable Linear Advance
changing from 12V to 24V
tried Different Hotends and Extruders
rewired everything
and lastly played with settings in Marlin without knowing what they do

Now I dont know what to do and looking for help. The issue is consistant so I think this might be a software problem.

I would like to hear your opinion and how to tackle the problem again. Maybe my thoughts went in the wrong direction.
I will try to answer as soon as possible but university is busy at the moment.

Thanks

-Emile

 **Files**

[Hypercube issue.zip](https://github.com/MarlinFirmware/Marlin/files/3453769/Hypercube.issue.zip)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Steps to Reproduce

1. I built two printers both have the issue. I followed many the instructions on Youtube how to set up Marlin. So I dont realy know how to reproduce the issue. I did everything the same.

**Expected behavior:** [prints like some other hypercubes with a good quality... hopefully]

**Actual behavior:** [Gaps in the infill, strange lines at the end of a linear motion]

thinkyhead, 2019-07-31:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-07-31 by thinkyhead

mulski, 2019-07-31:
As I wrote... I tried getting help in varios communities, but non of them worked. This is the reason why I want to solve it here. As I wrote I tried everything but nothing worked. So I think it is the Software. And I am looking forward to solve it!

> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this [...]

marciot, 2019-08-01:
@thinkyhead: Emile has contacted me earlier and discussing the problem it seemed like this went beyond a mere support request. From what I understand, nobody in various forums have been able to get to the bottom of this and I also showed this to some of the engineers at Lulzbot and the only idea that came from that (coasting) was ruled out by Emile.

Anyhow, it seems like there is more to this than meets the eye.

mulski, 2019-08-01:
Well done! Thanks a lot!

Hope it helps.

-Emile

lrpirlet, 2019-08-04:
hummm… I have NOT any idea what has been discussed in the various forums… However, I have a strong feeling that this is NOT software... Software is 1 or 0, yes or no... only the hardware is able to react analogically rather than digitaly...

To me the imperfections seems to be taking place when acceleration takes place (acceleration may be positive or negative)...

That said, you have a core xy device. That means both motors are in rotation when the nozzle goes along the X or Y coordinates... The infill on the pictures are at 45° normally ONE MOTOR ONLY is then running... The problem seems to be, or rather is obvious, at the end of the move...

Maybe those questions were asked before. Have you tried to print the cube at 45° to the X (or Y) coordinate? If yes did you get the same pattern? Could you have a "springy" set of belts? what kind of tensionner do you use, could it make the belt become longer or/and shorter during acceleration/deceleration?

Last but not least, when facing such a nonsense problem, I try to think "out of the box". I write any and every stupid (or funny) idea before to prove they are wrong with experiments, observing from away to generate more unrelated ( I mean at first unrelated) ideas... Good luck.

lrpirlet, 2019-08-04:
@mulski... sorry, forgot to address you my thought… please read previous entry in GitHub…

mulski, 2019-08-04:
@lrpirlet Thanks for your answer. I am am thankfull of every thought. Youre right. The problem happens at the end of a move. I tried rotating the cube in the past, it didnt helped. I just tried it again to confirm it. 
i am using the springy type of GT2 belts, but got the pretty tight. I pull them with pliers and screw the screws of the belt holder down. 
I played with acceleration an speed in the past. Currently I am printig with 30mm/s at 1000mm/ss.

Why do you think the problem is related to the movement of the other two motors? I think it might be the extruder.

The print is rotated 45°.
[image]

thinkyhead, 2019-08-05:
>  I think it is the Software

Sorry for the inconvenience. We close issues unless is it _proven_ to be the firmware and there is something _actionable_. Keep using the various support forums at your disposal until there is something actionable. We are not extra smart here, and it takes time and effort to go hunting down mysterious issues, especially without any good leads. Until there are good leads, it's impossible for us to even begin.

lrpirlet, 2019-08-05:
@mulski 
OK, I will not chatter too much on a closed call... However I still want to answer your question.

> Why do you think the problem is related to the movement of the other two motors? I think it might be the extruder.

_Well, I think out of the box.._. YES I see that the extruder stopped too early... BUT what if the move was impeded in some way when both A and B motor were running in such a way that the extruder stopped pushing material when it thought that the nozzel was at the end of the line while in fact it still needed to go 2 more milimeters ??? (sorry, I am tired and english is not my mother language...)

Now what can I see from that experiment?
With only one motor running we have the SAME pattern on the infill, we have the same pattern on the outside wall... This tend to say that the problem is NOT tied with the fact that we have a corexy device...

I take it for granted that you have a [right handed cartesian](https://en.wikipedia.org/wiki/Cartesian_coordinate_system) device (follow the link and search for orientation)… do no make my mistake, when the platter goes down, the distance of the xyplane to the nozzle increase.. for the platter to go down the finger goes up ( the reference is the origin on the xy plane)….
In fact the Z on the cube makes me believe that the coordinate are right handed unless you need to take the mirror to get the z to print correctly

What the firmware people do ask is that you do some homework... I do know nothing about your configuration... 
If it was me, I would SIMPLIFY it until it work better...
For example, if you have mesh levelling, make sure to disable levelling fade height and compare before after... or if you use linear advance set K factor to zero

good luck

mulski, 2019-08-06:
> Well, I think out of the box... YES I see that the extruder stopped too early... BUT what if the move was impeded in some way when both A and B motor were running in such a way that the extruder [...]
Your thought is very interesting. And I would like to go deeper into this. What is your motherlanguage. Im from Germany. Maybe this makes it easier.

> I take it for granted that you have a right handed cartesian device (follow the link and search for orientation)… do no make my mistake, when the platter goes down, the distance of the xyplane to [...]
In fact the Z on the cube makes me believe that the coordinate are right handed unless you need to take the mirror to get the z to print correctly

I dont think my printer uses the right hand rule. I would say it is the lefthand rule ;-) the origin is on the left front corner. And x goes from left to right. Y from front to Back.

I will  setup e new Marlin.