MarlinFirmware/Marlin #14811: Dual X carriage

hallphilip, 2019-08-02:
When I enable Dual X Carriage neither carriage will home and pops up a homing error. If I disable the feature everything homes just fine. any help with this would be great. I am willing to test anything. this was using the latest version on github.

//file Configuration.h Release:origin/bugfix-2.0.x
//section info
#define STRING_CONFIG_H_AUTHOR "Phil Hall_DPE_V1.2"
//section machine
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
#define MOTHERBOARD BOARD_BIGTREE_SKR_PRO_V1_1
#define CUSTOM_MACHINE_NAME "DPE-BW"
#define INVERT_X_DIR true
#define INVERT_Y_DIR false
#define X_BED_SIZE 273
#define Y_BED_SIZE 250
#define Z_MAX_POS 250
//#define MIN_SOFTWARE_ENDSTOP_X 
//#define MIN_SOFTWARE_ENDSTOP_Z 
//section extruder
#define EXTRUDERS 2
#define PARKING_EXTRUDER_PARKING_X { -71, 273 }
#define PARKING_EXTRUDER_GRAB_DISTANCE 0
//#define PARKING_EXTRUDER_SOLENOIDS_INVERT 
#define PARKING_EXTRUDER_SOLENOIDS_DELAY 600
#define SWITCHING_TOOLHEAD_X_SECURITY -35
//#define HOTEND_OFFSET_X {0.0, 71.00}
//#define HOTEND_OFFSET_Y {0.0, 5.00}
#define HOTEND_OFFSET_Z { 0.0, 0.8 }
//#define THERMAL_PROTECTION_CHAMBER 
#define DISABLE_INACTIVE_EXTRUDER true
#define INVERT_E0_DIR true
#define INVERT_E1_DIR true
//section temperature
#define TEMP_SENSOR_0 20
#define TEMP_SENSOR_1 20
#define TEMP_SENSOR_BED 11
#define TEMP_RESIDENCY_TIME 6
#define TEMP_HYSTERESIS 5
#define TEMP_BED_HYSTERESIS 5
#define PID_EDIT_MENU 
#define PID_AUTOTUNE_MENU 
#define PID_PARAMS_PER_HOTEND 
#define PID_FUNCTIONAL_RANGE 15
#define DEFAULT_Kp.0 15.98
#define DEFAULT_Ki.0 2.87
#define DEFAULT_Kd.0 22.25
#define PIDTEMPBED 
#define DEFAULT_bedKp.0 278.84
#define DEFAULT_bedKi.0 54.55
#define DEFAULT_bedKd.0 356.56
#define NOZZLE_CLEAN_END_POINT {100, 60, (Z_MIN_POS   1)}
//section homing
#define USE_XMAX_PLUG 
//#define ENDSTOPPULLUPS 
#define ENDSTOPPULLUP_XMAX 
#define ENDSTOPPULLUP_XMIN 
#define ENDSTOPPULLUP_YMIN 
#define ENDSTOPPULLUP_ZMIN 
#define X_MIN_ENDSTOP_INVERTING true
#define Y_MIN_ENDSTOP_INVERTING true
#define Z_MIN_ENDSTOP_INVERTING true
#define X_MAX_ENDSTOP_INVERTING true
#define X_DRIVER_TYPE TMC2130
#define Y_DRIVER_TYPE TMC2130
#define Z_DRIVER_TYPE DRV8825
//#define X2_DRIVER_TYPE TMC2130
#define E0_DRIVER_TYPE DRV8825
#define E1_DRIVER_TYPE DRV8825
#define E2_DRIVER_TYPE TMC2130
//section motion
#define DISTINCT_E_FACTORS 
#define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 160, 3200, 1666.68 }
#define DEFAULT_MAX_FEEDRATE { 100, 100, 10, 60 }
#define DEFAULT_MAX_ACCELERATION { 900, 900, 100, 10000 }
#define DEFAULT_ACCELERATION 900
#define DEFAULT_RETRACT_ACCELERATION 900
#define DEFAULT_TRAVEL_ACCELERATION 900
//section probes
//#define Z_SERVO_ANGLES {70,0}
//section calibrate
//#define ENABLE_LEVELING_FADE_HEIGHT 
//#define SEGMENT_LEVELED_MOVES 
#define MESH_TEST_HOTEND_TEMP 205.0
#define MESH_TEST_BED_TEMP 60.0
#define MESH_INSET.0 10
//section lcd
#define REVERSE_ENCODER_DIRECTION 
#define INDIVIDUAL_AXIS_HOMING_MENU 
#define SPEAKER 
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 
#define U8GLIB_SH1106 

//file Configuration_adv.h Release:origin/bugfix-2.0.x
//section temperature
#define THERMAL_PROTECTION_PERIOD 240
#define THERMAL_PROTECTION_HYSTERESIS 10
#define WATCH_TEMP_PERIOD 180
#define THERMAL_PROTECTION_BED_PERIOD 240
#define THERMAL_PROTECTION_BED_HYSTERESIS 10
#define WATCH_BED_TEMP_PERIOD 180
#define WATCH_BED_TEMP_INCREASE 4
//#define AUTOTEMP 
//#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 2
//section extruder
#define LIN_ADVANCE 
#define LIN_ADVANCE_K 0
//section homing
#define X_HOME_BUMP_MM 0
#define Y_HOME_BUMP_MM 0
#define Z_HOME_BUMP_MM 4
#define HOME_Y_BEFORE_X 
//section extras
#define X_DUAL_ENDSTOPS 
#define DUAL_X_CARRIAGE 
#define DEFAULT_DUAL_X_CARRIAGE_MODE DXC_FULL_CONTROL_MODE
#define ADAPTIVE_STEP_SMOOTHING 
#define ARC_P_CIRCLES 
//#define MINIMUM_STEPPER_DIR_DELAY 20
#define RETRACT_LENGTH_SWAP 10
#define RETRACT_RECOVER_LENGTH 5
#define RETRACT_RECOVER_LENGTH_SWAP 10
#define SPEED_POWER_SLOPE.0 0.3922
#define SPEED_POWER_MIN.0 10
#define SPEED_POWER_MAX.0 100
#define USER_DESC_2 "Preheat for PLA"
#define USER_DESC_3 "Preheat for ABS"
//section machine
#define AXIS_RELATIVE_MODES {false, false, false, false}
//section lcd
#define MANUAL_FEEDRATE {50*60, 50*60, 4*60, 60}
#define ENCODER_10X_STEPS_PER_SEC 75
#define ENCODER_100X_STEPS_PER_SEC 160
#define BABYSTEPPING 
#define DOUBLECLICK_FOR_Z_BABYSTEPPING 
//section tmc
#define X_MICROSTEPS.0 128
#define X2_MICROSTEPS.0 128
#define Y_MICROSTEPS.0 128
#define Y2_MICROSTEPS.0 128
#define Z_MICROSTEPS.0 128
#define Z2_MICROSTEPS.0 128
#define Z3_MICROSTEPS.0 128
#define E0_MICROSTEPS.0 128
#define E1_MICROSTEPS.0 128
#define E2_MICROSTEPS.0 128
#define E3_MICROSTEPS.0 128
#define E4_MICROSTEPS.0 128
#define E5_MICROSTEPS.0 128
//section tmc_smart
#define X_CURRENT 900
#define Y_CURRENT 900
#define X_CS_PIN PA15
#define Y_CS_PIN PB8
//#define X2_CS_PIN PG12
#define E2_CS_PIN PG12
#define TMC_USE_SW_SPI 
#define TMC_SW_MOSI PC12
#define TMC_SW_MISO PC11
#define TMC_SW_SCK PC10
//#define STEALTHCHOP_Z 
//#define STEALTHCHOP_E 
#define CHOPPER_TIMING CHOPPER_DEFAULT_24V
#define MONITOR_DRIVER_STATUS 
//#define X2_HYBRID_THRESHOLD 100
//#define Y2_HYBRID_THRESHOLD 100
//#define Z_HYBRID_THRESHOLD 3
//#define Z2_HYBRID_THRESHOLD 3
//#define Z3_HYBRID_THRESHOLD 3
//#define E0_HYBRID_THRESHOLD 30
//#define E1_HYBRID_THRESHOLD 30
#define E2_HYBRID_THRESHOLD 100
//#define E3_HYBRID_THRESHOLD 30
//#define E4_HYBRID_THRESHOLD 30
//#define E5_HYBRID_THRESHOLD 30
#define SENSORLESS_HOMING 
#define X_STALL_SENSITIVITY 5
#define Y_STALL_SENSITIVITY 5
#define TMC_DEBUG

InsanityAutomation, 2019-08-02:
Turn off x dual endstops. Make sure use_xmin_plug is enabled. Attach the whole config file next time, its easier to look for things in the file where we know what matters than it is to look through a list of changes like this. Also likelihood of something important getting missed like this is high.

hallphilip, 2019-08-03:
I'll Try that out, and thanks for the advice on future post.

hallphilip, 2019-08-03:
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020000

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// config/examples/SCARA and customize for your machine.
//

[... 2102 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// Delay (in milliseconds) before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES

hallphilip, 2019-08-03:
That still didn't help, I have no movement on E2, E1 won't home but I can move it manually. I noticed I also can't seem to get the Steps per mm set to use all my values using distinct E values.

Roxy-3D, 2019-08-03:
@hallphilip    Did you do a M502 and M500 after loading the firmware?     You can either .ZIP up the files and attach them.   Or change the .h to .txt to attach them.

hallphilip, 2019-08-03:
@Roxy-3D  thanks for the reply. I have not tried M500 or M502 yet because the SKR pro has no EEPROM, yet anyhow. Could that be an issue? Thanks for the info about the file attachment. My first attempt turned into a failure.

hallphilip, 2019-08-03:
Is there any other info I can add to help solve this?

Roxy-3D, 2019-08-03:
.ZIP up both the Configuration.h and the Configuration_adv.h files and post them.   The IDEX setup stuff is in Configuration_adv.h.   So that is the most interesting.

Not having EEPROM should not be a problem.   But the reason I was asking was there are a some IDEX setup numbers that get pulled from the EEPROM (if it is present).   For example, the parking location and the X & Y offsets.

hallphilip, 2019-08-03:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3464316/Configuration.zip)
Thanks for the help.

Roxy-3D, 2019-08-03:
This is an IDEX (Independent Dual Extruder) machine, right?

If so...   To turn on the IDEX features, you need to enable:

#define DUAL_X_CARRIAGE

hallphilip, 2019-08-03:
It was turned on I had posted the wrong config. I had two on hand so I could still print with T0.

hallphilip, 2019-08-03:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3464361/Configuration.zip)
 here is the correct one

hallphilip, 2019-08-03:
I found a hardware issue with my E2 driver.

closed 2019-08-04 by hallphilip

InsanityAutomation, 2019-08-04:
Sounds good! Thanks for updating.