MarlinFirmware/Marlin #14812: Enabling Panelolu support leads to compiler errors

criede, 2019-08-02:
<
### Description

Marlin does not compile sucessfully with LCD_I2C_PANELOLU2 support

### Steps to Reproduce

1. git clone/git pull of 2.0.x bugfix release on Aug 2 2019
2. Configure Marlin for RAMPS board in the usual way. I also experienced the same error when configuring for an Archim2 board.
3. Enable LCD_I2C_PANELOLU2
3. Compile

**Expected behavior:** Compiler finishes with no errors

**Actual behavior:** I get:
In file included from sketch/src/gcode/calibrate/G28.cpp:50:0:
sketch/src/gcode/calibrate/../../lcd/ultralcd.h: In static member function 'static void MarlinUI::buzz(long int, uint16_t)':
sketch/src/gcode/calibrate/../../lcd/ultralcd.h:264:9: error: 'lcd' was not declared in this scope
         lcd.buzz(duration, freq);
         ^
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

criede, 2019-08-02:
Relation to https://github.com/MarlinFirmware/Marlin/commit/c4bb458763c810f29025fc6b2c5829022ed509c7 ?

Ludy87, 2019-08-02:
Try this PR https://github.com/MarlinFirmware/Marlin/pull/14803

boelle, 2019-08-06:
@criede did @Ludy87's suggestion help?

criede, 2019-08-06:
Just did a 
  git clone https://github.com/Ludy87/Marlin.git Marlin-Ludy87
, the open it with Arduino, remove comment in front of #define LCD_I2C_PANELOLU2 and compiled it.
Result: Same error:

In file included from sketch/src/module/tool_change.cpp:78:0:
sketch/src/module/../lcd/ultralcd.h: In static member function 'static void MarlinUI::buzz(long int, uint16_t)':
sketch/src/module/../lcd/ultralcd.h:243:7: error: 'lcd' was not declared in this scope
       lcd.buzz(duration, freq);
       ^
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.

Ludy87, 2019-08-06:
Wrong tree https://github.com/Ludy87/Marlin/tree/bf2_buzzer2

branche: bf2_buzzer2

criede, 2019-08-06:
OK, getting better. 
  git clone --branch bf2_buzzer2 https://github.com/Ludy87/Marlin.git Marlin-Ludy87
New error:
sketch/src/lcd/HD44780/ultralcd_HD44780.cpp: In static member function 'static void MarlinUI::update_indicators()':
sketch/src/lcd/HD44780/ultralcd_HD44780.cpp:1100:26: error: 'class Temperature' has no member named 'degTargetBed'
       if (thermalManager.degTargetBed() > 0) leds |= LED_A;
                          ^
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.

criede, 2019-08-06:
#define TEMP_SENSOR_BED 1
(default is 0)
Then:

/home/chr/Arduino/libraries/arduino-display-lcdkeypad/LcdKeypad.cpp:18:1: error: expected class-name before '{' token
 {
 ^
/home/chr/Arduino/libraries/arduino-display-lcdkeypad/LcdKeypad.cpp: In constructor 'LcdKeypad::LcdKeypad(LcdKeypad::MCPType, uint8_t, uint8_t, uint8_t, int, int, int, int, int, int, int, bool)':
/home/chr/Arduino/libraries/arduino-display-lcdkeypad/LcdKeypad.cpp:63:59: error: 'IS_RECURRING' is not a member of 'Timer'
 , m_keyPollTimer(new Timer(new KeyPollTimerAdapter(this), Timer::IS_RECURRING, s_defaultKeyPollTime))
                                                           ^
/home/chr/Arduino/libraries/arduino-display-lcdkeypad/LcdKeypad.cpp: In destructor 'virtual LcdKeypad::~LcdKeypad()':
/home/chr/Arduino/libraries/arduino-display-lcdkeypad/LcdKeypad.cpp:113:26: error: 'class Timer' has no member named 'adapter'
   delete m_keyPollTimer->adapter();
                          ^
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.

Ludy87, 2019-08-06:
Deine Marlin Version entspricht der die im Branche ist! Nutze den Branche!

criede, 2019-08-06:
Ich bin mir gerade nicht klar, was hier die Erwartungshaltung mir gegenüber ist. Ich habe einen Fehler gemeldet. Der Fehler ist klar und mit minimalen Aufwand nachvollziehbar. Soweit ich das einschätzen kann, wurde da beim Aufräumen was verschlimmbessert. Und das sollte derjenige, der aufgeräumt hat, problemlos wieder hinkriegen.
Aktuell verliere ich mich in Branches und bekomme immer neue Fehlermeldungen. 
Ludy, tu mir einen Gefallen. Nimm den Code, aktiviere Panelolu und schau selbst was passiert. 
Mir reicht mir das wissen, das dran gearbeitet wird und zu gegebener Zeit eine kurze Meldung "fixed in 2.0.x" völlig. 
Danke - ich weiß Eure Arbeit zu schätzen.

Ludy87, 2019-08-06:
Schau in meine angehängte zip und du wirst DEINE Config vorfinden und ggf. den Fix, den du bitte Testen sollst beinhalten. Solange der Fix nicht bestätigt ist, wird er auch nicht veröffentlicht!

criede, 2019-08-06:
Zu sagen was man will steigert die Change zu bekommen was man will.... ;-)

extracted the ZIp, compiling it now with Arduino 1.8.9 auf Linux. Error message: 

/home/chr/Arduino/libraries/arduino-display-lcdkeypad/LcdKeypad.cpp:18:1: error: expected class-name before '{' token
 {
 ^
/home/chr/Arduino/libraries/arduino-display-lcdkeypad/LcdKeypad.cpp: In constructor 'LcdKeypad::LcdKeypad(LcdKeypad::MCPType, uint8_t, uint8_t, uint8_t, int, int, int, int, int, int, int, bool)':
/home/chr/Arduino/libraries/arduino-display-lcdkeypad/LcdKeypad.cpp:63:59: error: 'IS_RECURRING' is not a member of 'Timer'
 , m_keyPollTimer(new Timer(new KeyPollTimerAdapter(this), Timer::IS_RECURRING, s_defaultKeyPollTime))
                                                           ^
/home/chr/Arduino/libraries/arduino-display-lcdkeypad/LcdKeypad.cpp: In destructor 'virtual LcdKeypad::~LcdKeypad()':
/home/chr/Arduino/libraries/arduino-display-lcdkeypad/LcdKeypad.cpp:113:26: error: 'class Timer' has no member named 'adapter'
   delete m_keyPollTimer->adapter();
                          ^
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.

[image]

Ludy87, 2019-08-06:
Das ist jetzt kein Marlin Fehler mehr, scheinbar hast du die falsche lib installiert bzw. garkeine

criede, 2019-08-06:
OK, das ist was anderes. Ich geh mal suchen.

criede, 2019-08-06:
ok, in arduino, you can add https://github.com/dniklaus/arduino-display-lcdkeypad from the "manage libraries" menu. That doesn't work. 
Adding https://github.com/lincomatic/LiquidTWI2/archive/master.zip as ZIP archive works. 
Thanks for your help!

Ludy87, 2019-08-06:
Thank you for your feedback!

Has the function been preserved or are there problems with the display?

Ludy87, 2019-08-15:
@thinkyhead, can this (https://github.com/Ludy87/Marlin/tree/bf2_buzzer2) be seen as a solution to the problem? 

Here could be closed.

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle

chepo92, 2019-11-03:
@Ludy87  your fix is working for Viki 1 I2C LCD, had to make some minor additions, should PR them to your branch?