MarlinFirmware/Marlin #14813: [FR] Reworked LCD Menu layout/tree.

AmyTheCute, 2019-08-02:
### Description

Reworked Menus for LCD Screens

#### Additional Information

I think right now the menu's are very cumbersome and hard to remember and hard to use.
Why don't we instead of control + tune/prepare have a Settings menu where you can control things like Linear advance acceleration jerk etc and another menu for Fan speed (which I don't understand why it isn't in percentage like everywhere else?) and moving axis.

I'd also love a Load/Unload filament option or eject filament that doesn't need you to use the knob to set a distance like the other axis. or something you keep holding the button on to push the axis in/out.

I think the prusa's menu's are a good example!

I think holding the button could also be used in some parts of the menu for faster more comfortable usage. For example maybe holding the button would set the value to zero in the temperature section?

InsanityAutomation, 2019-08-02:
Have you used Marlin 2.0 recently? You're describing stuff that changed almost a year ago.

AmyTheCute, 2019-08-02:
Oh it's you from the the server! are you sure it's different in 2.0?! A quick googling made me believe it's the same. I should try that then.

InsanityAutomation, 2019-08-02:
Yup, im everywhere! 2.0 had a complete reorganization thanks in large part to Marcio then code cleanup mostly by thinkyhead so its drastically different.

AmyTheCute, 2019-08-02:
I'll try it later after my print then! It's scary to migrate but I can manage it! hopefully...
I really wish I could contribute to marlin like you all but I'm not really good enough at coding to understand complex projects like this!

InsanityAutomation, 2019-08-02:
This is just the product of slightly more monkeys banging randomly on keyboards than it took to make the works of Shakespeare... :-D

closed 2019-08-02 by AmyTheCute