MarlinFirmware/Marlin #14814: [BUG?] Filament Change not unloading when not actively printing

michaelsheridan23, 2019-08-02:
In Marlin 1.1.9:

After enabling the advanced park feature and using all of the default setting (I have a direct drive extruder) the printer will not unload the filament when using the LCD menu while the printer is idle and not actively running a print job. When the "unload filament" menu item is selected the printer prompts for the temperature setting, then after selecting it displays a please wait as the appropriate temperature is reached, after the extruder is to temperature the head moves up in Z then performs the "purge" sequence (retract, pause, ram/purge) and then moves the head back down to the original location and the LCD returns to the main screen. 

What I expected to happen was after the "purge"  sequence the filament would be unloaded and then the system would wait for the user to remove the filament and then using the LCD (by pushing a button) acknowledge that the filament has been removed, then the sequence would end and the LCD would return to the home screen. 

Do I not have something not set up properly? Or does this work in Marlin 2.0 as I am thinking? Or does this feature not operate as I think it should and I am just mistaken on how this is supposed to work?

[Marlin 1.1.9 Advance Park Issue Files.zip](https://github.com/MarlinFirmware/Marlin/files/3462292/Marlin.1.1.9.Advance.Park.Issue.Files.zip)

robbycandra, 2019-08-03:
Go to marlin 2.0

boelle, 2019-08-06:
yep, only bugfix 2.0.x is updated

@michaelsheridan23 does it work in 2.0.x ?

pseudex, 2019-08-08:
You can check the max extrusion length in the config file. I had a problem were there were exactly the same and it didn't do the unload. Sanity checks will prevent that at compile time. But somehow this happened when I changed config vales on the LCD.
I use this feature a lot on marlin 2.0 and it's working great.

michaelsheridan23, 2019-08-15:
@boelle After updating to 2.0.x it does work as expected. Thank for the reply.

boelle, 2019-08-15:
then i will close this one, we can always reopen if it does not work in 2.0.x

closed 2019-08-15 by boelle