MarlinFirmware/Marlin #14818: [BUG] SKR Mini E3 & E3 DIP - Temp Error if USB connected on boot

reloxx13, 2019-08-02:
### Description

If USB is plugged in (pi3b+ with octoprint) and i power on the printer and pi, the printer will halt on a temp error.

Occurs with SKR mini E3 and E3 DIP.
Check other both issues at the bottom.

With the suggested workaround 

`#define BOGUS_TEMPERATURE_GRACE_PERIOD 2000`

The printer shows 72°C E0 and -14°C HB after boot with USB

I tried multiple reboots.

After unplugging USB and rebooting, the temp readings are okay.
Plugging USB in now, the temp settings stay okay.

#### Additional Information

config: https://github.com/reloxx13/Marlin/tree/my-skr-mini-e3/Marlin

Issues from Bigtreetech
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/issues/2
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/4

lovethat777, 2019-08-03:
Can confirm, had to set THERMAL_PROTECTION_GRACE_PERIOD to 6000 (ms)

thinkyhead, 2019-08-05:
@p3p — I wonder if we can do something to get readings stabilizing or the ADCs initialized in a safer manner, and in some way get temperature readings started before SD initialization, even in those cases where settings need to be fetched from SD. I'm not sure the latency improvement can do anything about the readings we get on bootup being equivalent to a thermal runaway.

In the meantime we could try add some logging to the temperature reading and SD init routines and see what's going on in detail during these long boot-ups.

reloxx13, 2019-08-05:
Wouldnt it be enough to update the temp reading after boot?   
It just sits there with the wrong temp on the LCD.   
   
When i connect USB after it booted, in octoprint i can see the correct temp readings.   
But im not sure now if i could start a print, sometimes it get bugged and wont execute any command, but it shows "connected".

p3p, 2019-08-05:
@thinkyhead I can't really comment on whats going on with the STM boards, I don't know if they use any filtering (in addition to Marlins) like the LPC176x , as far as I can tell since adjusting the LPF the LPC176x doesn't have any issues at boot.

opcow, 2019-08-08:
Just installed an SKR E3 DIP. My bogus temps are on the hotend and bed. I see 298C on the HE using a USB cable with no 5v line, -14C on the HE or both heaters if powered. The workaround of using BOGUS_TEMPERATURE_GRACE_PERIOD doesn't help since the bogus temp is never resolves.

lovethat777, 2019-08-08:
Are you using the lastest 2.0 bugfix release, if yes, BOGUS_TEMPERATURE_GRACE_PERIOD was [just removed](https://github.com/MarlinFirmware/Marlin/commit/5bc2fb022ca437d7e70f195bdb7d946ceb3a623a)

tomoinn, 2019-08-27:
This is fixed in the current upstream marlin, can confirm it's now not a problem :)

closed 2019-08-28 by reloxx13