MarlinFirmware/Marlin #14819: CoreXY

DarkAlchy, 2019-08-02:
I have no issues with the old Marlin on an 8 bot Melzi 2.0 with my TronXY X5S CoreXY printer but I am having several issues I am working through and this has me stumped.

I tell it to move X, or Y at my standard 150mm/s speed and it gets mid way across and groans as if something is dragging but by hand I can do over 300mm/s and it is smooth as butter and no sounds nor was there an issue on the older Marlin.

Was anything changed in the CoreXY parts that may be causing this?  I have tried with stealthchop and without and it is the same thing.

I am seriously stumped because if I hook the old board back up it runs again without any issues.

lovethat777, 2019-08-03:
I've just set up a corexy system with Marlin 2.0 without problem.

Enable TMC_DEBUG, play with your axis and then issue a M911 ("Report stepper driver overtemperature pre-warn condition"), what's the output ?

Do you have HYBRID_THRESHOLD defined in your configuration_adv ?

DarkAlchy, 2019-08-03:
I do not have hybrid_threshold defined as I am trying to start with as few options on as possible then branch out from there.  I have not played with the M911 yet and one odd things is that my X and Y steps per mm had to be over halved from what they were yet all things are equal and Z was still spot on and E I have not tested yet to see if it is off or not.

So far the reduction in steps per mm has taken care of a lot of problems.

There was another issue I had but not sure if I still do but I was watching this video - https://www.youtube.com/watch?v=HOryknQnOd4 since it his creation is a CoreXY as well so I manually had to invert some things as he showed but when my gcode file kicked in my nozzle went to the direct diagonal side from where it should have been so that confused me as well.

DarkAlchy, 2019-08-03:
With the changes I mentioned above it all works now and the SD card issue I had I will check in a few minutes and if still present will open an issue for it but this issue seems to now be taken care of so I am closing this.

Thank you for the help.

closed 2019-08-03 by DarkAlchy