MarlinFirmware/Marlin #14820: [BUG] Cannot initiate a new print

lovethat777, 2019-08-03:
### Description

Sometimes, after a successful print from SD, another print cannot be initiated, as the print confirm menu does nothing when Print is selected.

### Steps to Reproduce

1. Print any file from SD
2. Wait for completion
3. Try to print the same file from SD

**Expected behavior:** Another print should be run

**Actual behavior:** Cannot go past the print confirm menu

#### Additional Information

[Conf.zip](https://github.com/MarlinFirmware/Marlin/files/3463228/Conf.zip)

boelle, 2019-09-24:
@lovethat777 is the problem still there?

lovethat777, 2019-09-24:
I will have to test. Has someone corrected the issue?

boelle, 2019-09-24:
2.0.x is updated almost every day so its a big change its fixed

lovethat777, 2019-09-24:
Do you have a commit # regarding this particular issue?

boelle, 2019-09-24:
no i thought you had that ?

lovethat777, 2019-09-24:
You say it's updated everyday. Do you have any information this very bug has been addressed, such as a commit which mentions correcting this bug?

boelle, 2019-09-24:
no i assume that since you have the issue then you looked at the commits on a daily basis so that you knew if the issue is fixed

since i dont have the issue i dont have a reason to look every single commit

lovethat777, 2019-09-24:
Unfortunately I haven't had the time to do so. I will keep everyone posted if I do update it in the near future. My setup is working pretty well except for this minor bug. In case anyone has info on the subject, they are welcome to share it.

boelle, 2019-10-21:
will test this one with code flashed 3 days ago (commit 725ec68fa) print takes 2½ hours

boelle, 2019-10-21:
i can not reproduce the error, i can start the same print again once the first one is completed

given that there has been no activity i will close this one as fixed

we can always reopen if it becomes confirmed that there is an issue

closed 2019-10-21 by boelle