MarlinFirmware/Marlin #14822: During a print job Z does not engage.

DarkAlchy, 2019-08-03:
CoreXY (TronXT X5S) and before a print job I can easily G1 anywhere with Z but as soon as a print job begins Z disengages and never lowers with each layer.  I tried this with pronterface as a well as via the sd card and when I start the print I can move Z by hand with ease when it should be locked/engaged and hard to move as it sits ready to take layer commands.

reloxx13, 2019-08-03:
upload ur configs

J-charles-C, 2019-08-03:
I,

With day's commit, no problem with my Hypercube (CoreXY) and Re-Arm Motherboard.
best

DarkAlchy, 2019-08-03:
I firmly believe, and I am about to test this, that the two issues I have up are interelated because I use 4800 steps/mm for Z.

DarkAlchy, 2019-08-03:
Yep, I just tested and the 4800 steps/mm is the culprit and when I go to print nothing but I changed it to 2400 and Z was engaged and moved as it should.

sjasonsmith, 2019-08-03:
I rebased my firmware last night to make sure I'm using the latest changes. While I've confirmed the issue configuring values > 3276 through the LCD, I've also verified that I can print fine with 6400 Z steps/mm when I configure it through the Configuration.h (with an EEPROM reset) or using the M92 command.

Either you don't have the right value saved (a stale value in the EEPROM will overwrite anything you change in Configuration.h), or some other configuration issue is at play. As reloxx13 said, you would need to upload your configuration files for anyone to be able to help with that. I'd also be interested in the output from a M503 command when you have the issue reproducing.

DarkAlchy, 2019-08-03:
My EEPROM is off so I am only using configuration.h values.

DarkAlchy, 2019-08-03:
I don't know what is going on and I am flustered beyond all belief.

I just changed over from 16 microstepping to 8 and changed my steps per mm from 4800 to 2400 and when I home Z it goes up and hits the endstop and locks up.  At least 4800, and 16, I can home Z.

Here are my config files with the 16 and 4800 in them and I hope someone spots something because I can't get Z to work.

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3464396/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3464397/Configuration.txt)

sjasonsmith, 2019-08-03:
Can you try reducing the max Z speed?

I know you've said your hardware can handle 30 mm/s, but that would require steps at 144 kHz, just for your Z axis. I have no idea how much Marlin can keep up with in your configuration, but lowering the speed will be a good troubleshooting step.

When I look at the examples which use high step rates, they almost all use a max Z speed of 5 mm/s. That would be a good place to start, and it close to your homing speed (6.66 mm/s) which seems to be working.

Also, can you verify whether you are running from a 12 or 24 V power supply?

gloomyandy, 2019-08-03:
You might also want to set...
#define MINIMUM_STEPPER_PULSE 2
the default settings can cause problems with TMC drivers with SKR boards.

DarkAlchy, 2019-08-03:
@gloomyandy That is a new one to me.

I switched out the driver for Z with a spare I had and it worked and 1/2 stepping while the other would not print.  Could that be because of that PULSE?

DarkAlchy, 2019-08-03:
I am on 12v and I am down to 10mm/s on Z.

sjasonsmith, 2019-08-04:
Did you increase the pulse, as @gloomyandy suggested, or just change the speed? I'm wondering whether your speed could be cranked back up with the pulse increased.

Also, you have INTERPOLATE enabled for your drivers, so I suspect you won't hear much difference between 1/8 or 1/16 microstepping. Both should get interpolated to 1/256 on the motor side.

DarkAlchy, 2019-08-04:
Interpolate I just turned on and is much quieter but not as quiet as it would be as more steps.  The speed I reduced and all of this is before gloomy's response.  Right now I am fighting with Amazon over this driver (on chat now with them)  and looks like I have to return all of this so when done I will install gloomy's suggestion and report back.

DarkAlchy, 2019-08-04:
I can't tame Z.  I have tried gloomy's suggestion and suddenly everything stopped work.

I don't get it because it homes and just plain locks up as soon as it hits the home button and that is where it stays.  Steps no longer matters nothing matters that I have tried yet X, Y, E0 works.

DarkAlchy, 2019-08-04:
Question but if I have this #define DEFAULT_MAX_FEEDRATE          { 500, 500, 5, 50 }  and I do a G1 Zxx F6000 shouldn't that 5 (300mm/m) make the F6000 into a F300?  I know in Marlin 1.x it did.

DarkAlchy, 2019-08-04:
Something weird but I used to use Marlin from beta through 1.1.8 and setting the max feedrate was a global override but no longer.

Max feedrate
This is an important setting, it limits the seed of an axis. If the speed is to high,
the stepper motor will 'loose' steps, and your print will fail. The default is:
#define DEFAULT_MAX_FEEDRATE {1000, 1000, 500, 500} // (mm/sec)
For most printer the value 500 for the z axis is much to high, 5 is probably a
better choice.

I have - #define DEFAULT_MAX_FEEDRATE { 500, 500, 5, 50 } and if I do a G1 Zxx F6000 it stalls out the motor because it was too fast (you can hear it whine but not move) so I wonder why the global is no longer working as it used to?

DarkAlchy, 2019-08-04:
As a final gasp I went into the pins file and made E1 my Z to test if there might have been a bad pin or solder joint or port, etc... and exactly the same outcome.  I have done all of the testing that I can, and I have tried all avenues with complete failure so nothing left that I can do for now.  I might not be able to use this SKR until Repetier catches up or Marlin 2.x does.  Not sure why it is just me but I am used to it.

Thank you to everyone who attempted to help but I am leaving this issue up since it has not be fixed.

thinkyhead, 2019-08-05:
> For most printer the value 500 for the z axis is much to high, 5 is probably a

Indeed. Only the DELTA configurations use such high rates.

closed 2019-08-05 by thinkyhead

DarkAlchy, 2019-08-05:
This issue is in Marlin and has not been solved so why the closing of it?