MarlinFirmware/Marlin #14825: [BUG] BLTouch mesh incorrectly applied using Marlin 2.0

lovethat777, 2019-08-03:
### Description

I'm using a BLTouch clone (see below for repeatability test) with an already leveled bed with AUTO_BED_LEVELING_UBL.

I've slowed Z_PROBE_SPEED_SLOW to (Z_PROBE_SPEED_FAST / 4) and increased the probing count at each point MULTIPLE_PROBING 3, and am using a 5x5 mesh.

When printing a single layer calibration square, what I see is the Z increasing way too much (more than a millimeter) in the corner Xmax. Other corners seem fine.

I've already set up several machines with BLTouches and clones on Marlin 1 and never had an issue, until now.

I'm wondering if the issue lays with UBL_Z_RAISE_WHEN_OFF_MESH, which is disabled in my configuration.

My Zoffset is: M851 Z-1.90
Board: SKR 1.3

> M48 P10 X100 Y100 V2
Finished!
Mean: -0.022750 Min: -0.040 Max: 0.002 Range: 0.041
Standard Deviation: 0.013331

> G29 T
Bed Topography Report:
    ( 40,100)                      (110,100)
        0       1       2       3       4
 4 | -0.132  -0.142  -0.152  -0.139  -0.148
   |
 3 | -0.090  -0.087  -0.090  -0.094  -0.118
   |
 2 | -0.032  -0.017  -0.069  -0.032  -0.079
   |
 1 | -0.001  -0.027  -0.026  -0.059  -0.020
   |
 0 | -0.013  +0.011  -0.034  -0.017  -0.034
        0       1       2       3       4
    ( 40, 40)                      (110, 40)

### Steps to Reproduce

Setting Zoffset:

G28
M211 S0
Using LCD to lower Z height, using a sheet of paper as spacing
M851 Z-1.90
M211 S1
M500

Probing:

G28
G29 P1
G29 S1
M500

**Expected behavior:** Z correction according to the probing mesh

**Actual behavior:** Invalid Z correction

#### Additional Information

[Conf.zip](https://github.com/MarlinFirmware/Marlin/files/3464183/Conf.zip)

shitcreek, 2019-08-07:
There is mention that AUTO_REPORT_TEMPERATURE may interfere with UBL here: https://github.com/MarlinFirmware/Marlin/issues/7021
Have you tried G26? http://marlinfw.org/docs/gcode/G026.html
It is recommend to heat up the bed when probing as it will produce different results as oppose to unheated.
Incase you haven't already been here: http://marlinfw.org/docs/configuration/probes.html

lovethat777, 2019-08-07:
1. I indeed have AUTO_REPORT_TEMPERATURE defined. Does the referenced issue also apply to Marlin 2.0 ?
2. I'm currently upgrading my cloned metal hotend to a genuine microswiss one. Will report on the G26 once received and installed
3. I always probe the bed once well heated to the target temperature (80°C)
4. The probe itself works and is wired to Zmin, no problem there

For now I tried UBL and bilinear and obtain the same problematic results. Will update when I receive the new hotend.

shitcreek, 2019-08-08:
1. Not sure, but seeing how no commit was referenced to the issue, I think it should still apply.
3. 80* C is really high, I only go up to 50.

Just be sure to only have one option enabled at a time.

lovethat777, 2019-08-12:
When printing PETG, I use a bare spring steel sheet without any glue at 80°C. When the bed cools, parts separate themselves. You should try it.

I disabled AUTO_REPORT_TEMPERATURE and I still have the same issue.

I've noticed, when issuing a M420, that the bed leveling is off:

> echo:Bed Leveling OFF
echo:Fade Height OFF

However, when I do a M420 V1, the map is there:

> Bilinear Leveling Grid:
      0      1      2      3      4
 0 +0.530 +0.300 +0.215 +0.172 +0.108
 1 +0.455 +0.231 +0.116 +0.084 +0.033
 2 +0.387 +0.197 +0.033 -0.019 -0.055
 3 +0.300 +0.161 +0.016 -0.076 -0.155
 4 +0.164 +0.053 -0.044 -0.158 -0.279

I have RESTORE_LEVELING_AFTER_G28 defined. When a G28 is issued, I still have:

> echo:Bed Leveling OFF
echo:Fade Height OFF

I suppose this is a bug?

ManuelMcLure, 2019-08-12:
Did you actually activate the leveling with either ``M420 S1`` or ``G29 A1``? You can have a mesh loaded without having UBL active.

lovethat777, 2019-08-12:
I did. Somehow this setting was lost. I'll investigate further and report back.

closed 2019-08-26 by lovethat777