MarlinFirmware/Marlin #14826: Cant Compile for Fysetc Cheetah

ModMike, 2019-08-03:
### Description

Get error when trying to compile for Cheetah Fysetc Cheeth

### Steps to Reproduce

1) Clone latest Marlin Github 2.0.bugfix
2) Replace configs with Fysetc->Cheetah->Base config files
3) Compile using enc:fysetc_STM32F1

**Expected behavior:**

Compiles

**Actual behavior:**

`TypeError: unsupported operand type(s) for +: 'NoneType' and 'str':
  File "/Users/Michael/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 134:
    env.SConscript(item, exports="env")
  File "/Users/Michael/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/Michael/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "/Users/Michael/Desktop/Marlin-bugfix-2.0.x/buildroot/share/PlatformIO/scripts/fysetc_STM32F1.py", line 32:
    UPLOADER=platform.get_package_dir("tool-stm32duino") + '/stm32flash/stm32flash',`
#### Additional Information

I did check to make sure I had the latest merge for the python script and all good. Running on OSX.

RudyBenner, 2019-08-03:
There is a very recent update.

ModMike, 2019-08-03:
How recent?

RudyBenner, 2019-08-04:
[image]

RudyBenner, 2019-08-04:
not relevant.

ModMike, 2019-08-04:
My issue seems to be a directory issue.

ModMike, 2019-08-04:
Still no luck, same error after uninstalling and reinstalling everything. I can get the Fysetc one to build and am printing with it now. Would much rather be on official branch so I guess it will have to wait.

By the way I have a Cheetah 1.1, not 1.2.

RudyBenner, 2019-08-04:
neither one compiles.

ModMike, 2019-08-04:
Mine does. Just add add

`platform_packages = tool-stm32duino`

to env:fysetc_STM32F1

I alos had to manually instal STM32flash manually.

RudyBenner, 2019-08-04:
Yup.

[image]

thinkyhead, 2019-08-05:
Any way we can automatically fetch and include STM32flash? When is it required? We can always add a pre-build script to fetch and install it, if needed.

thinkyhead, 2019-08-05:
From the OP it looks like the build is succeeding, but in post-processing the `arm-none-eabi-objcopy` command is getting bad arguments. Strange behavior.

tpruvot, 2019-08-05:
maybe some build folder with spaces ?

thinkyhead, 2019-08-05:
Spaces should matter not at all in 2019. It's up to tool developers to ensure paths with spaces are treated correctly.

AngryMikey, 2019-08-05:
I am able to build fine with these configs, but my root path has no spaces.

thinkyhead, 2019-08-05:
Could be a PlatformIO or platform bug.

tpruvot, 2019-08-05:
i know it should, but #14697

thinkyhead, 2019-08-08:
Guess I should apply the fix from that issue!

thinkyhead, 2019-08-08:
I've applied a patch to the python scripts, so give it another shot.

boelle, 2019-09-24:
@ModMike did the patch fixproblem?

ModMike, 2019-09-24:
I e been working with Fytsec branch so no idea. Printer working beyond perfect so had no cause to recompile.

Will give it a shot this weekend and advise.

boelle, 2019-10-30:
@ModMike many moons have passed, did you get a chance to check it out?

ModMike, 2019-10-30:
Sorry but I did not. I was busy with work and I just recently got back into printing. Actually been going about 80 hours straight now!

I upgraded to Catalina and now I can't even load a boot loader! I will finish the job tonight and try to rebuild. By the way, the Fytsec controller is beyond perfect.

ModMike, 2019-10-30:
Ok just tried it and it built perfectly with only 4 warnings concerning unused definitions. Thanks!

`ompiling .pio/build/STM32F103RC_fysetc/lib112/U8glib-HAL_ID1932/clib/u8g_dev_ssd1306_128x64.c.o
.pio/libdeps/STM32F103RC_fysetc/U8glib-HAL_ID1932/src/clib/u8g_dev_ssd1306_128x32.c:153:22: warning: 'u8g_dev_ssd1306_128x32_univision_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x32_univision_init_seq[] PROGMEM = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/STM32F103RC_fysetc/U8glib-HAL_ID1932/src/clib/u8g_dev_ssd1306_128x32.c:83:22: warning: 'u8g_dev_ssd1306_128x32_adafruit2_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x32_adafruit2_init_seq[] PROGMEM = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/STM32F103RC_fysetc/U8glib-HAL_ID1932/src/clib/u8g_dev_ssd1306_128x32.c:48:22: warning: 'u8g_dev_ssd1306_128x32_adafruit1_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x32_adafruit1_init_seq[] PROGMEM = {`

boelle, 2019-10-30:
> By the way, the Fytsec controller is beyond perfect.

not being native english does this translate in to crap or the direct oposite?

ModMike, 2019-10-30:
It is VERY VERY good! I have a duet, a bunch of Panucatt boards and assorted controllers. This thing does it all with TMC2209 drivers for $27.00 USD. What a deal.

boelle, 2019-11-09:
will close this one as it seems to only throw warning and not do not fail

we can reopen if i'm wrong

closed 2019-11-09 by boelle