MarlinFirmware/Marlin #14828: [BUG?] SKR 1.3 and TMC2130 SPI communication failure

Dan018, 2019-08-03:
Using [this SKR 1.3](https://www.aliexpress.com/item/33010395838.html) board which shipped with the TMC2130 drivers and latest version (as of 03/08/19) bugfix 2.0.x. The drivers are labelled as "BIGTREETECH TMC2130 v3.0".

I've enabled software SPI, powered the board from a 24v PSU and setup config.h and config_adv.h as shown in the [BTT Youtube video for this board](https://www.youtube.com/watch?v=oaXfXkPYHpw&feature=youtu.be).

I'm confident that all configuration options are correct and the jumpers on the board are in the correct place.

Issuing an M122 gives:

**Driver registers:
		X	0xFF:FF:FF:FF	 Bad response!
		Y	0xFF:FF:FF:FF	 Bad response!
Testing X connection... Error: All HIGH
Testing Y connection... Error: All HIGH**

The motors will move in Pronterface but obviously the SPI communication isn't working and hence StallGuard doesn't work (the whole reason I'm trying to use 2130s).

I've checked the pins in the relevant pins.h file against the pinout on the BTT Github page for the board and everything is correct. As I have 5 of these drivers I've tried different ones in different locations, all with no success.

Is this a bug with this board, or are there any further checks I should do to test the SPI bus?

lovethat777, 2019-08-03:
I've got the exact same setup, except I don't use the stallguard feature of the 2130's yet. Using the very latest Marlin 2.0 bugfix, they work as expected on my end.

For the jumpers, each one marked in purple must be jumped:

[image]

Did you set the correct driver types ?

> #define X_DRIVER_TYPE  TMC2130
#define TMC_USE_SW_SPI
#define TMC_DEBUG
#define CHOPPER_TIMING CHOPPER_DEFAULT_24V

Dan018, 2019-08-03:
Ok, found the problem. Despite ordering SPI drivers, and the box saying SPI drivers, they were set up in step/dir mode requiring some annoyingly precise soldering. All working now.

closed 2019-08-03 by Dan018

thisiskeithb, 2019-08-04:
> Ok, found the problem. Despite ordering SPI drivers, and the box saying SPI drivers, they were set up in step/dir mode requiring some annoyingly precise soldering. All working now.

It's a common issue with these drivers being sold as "SPI" versions, unfortunately.

Talha909, 2019-12-07:
> Ok, found the problem. Despite ordering SPI drivers, and the box saying SPI drivers, they were set up in step/dir mode requiring some annoyingly precise soldering. All working now.

HI Dan, I have the same board with spi 2130. I am facing some problem . can you share the config and config adv

Glennvb85, 2019-12-07:
Had this exact same issue. I removed the SPI bridge and soldered the other two point. Added 2 pins for diagnostics but I still get the exact same issue, but now my steppers wont even move in pronter. Any idea whats happening?