MarlinFirmware/Marlin #14830: [BUG] Invalid characters displayed

lovethat777, 2019-08-03:
### Description

Marlin 2.0

When using REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, sometimes the "highlighted character" (i.e. the one being adjusted), for example when setting the bed temperature, is *sometimes* incorrect.** Link to the exact model: https://www.aliexpress.com/item/32823608306.html

My wiring seems correct and the LCD works as expected (except from this bug)

The default configuration is used:
> #define LCD_LANGUAGE en
#define DISPLAY_CHARSET_HD44780 JAPANESE

Example below. The text should be "+013.0".

[image]

If I use the knob back and forth, I can make it display "+013.0" as it should. The LCD has no dead pixel.

### Steps to Reproduce

1. Use the scoll whell when setting any value using the LCD

**Expected behavior:** The correct text should be displayed

**Actual behavior:** Invalid characters are sometimes displayed

#### Additional Information

[Conf.zip](https://github.com/MarlinFirmware/Marlin/files/3464358/Conf.zip)

thisiskeithb, 2019-08-04:
Have you tried shortening the cables/shielding them? The plain/common ribbon cables are prone to interference.

AnHardt, 2019-08-04:
In the upper half of the screen is still half of a "1" visible.
This is not a noise problem but a consequence of violating u8gLibs [picture loop](https://github.com/olikraus/u8glib/wiki/tpictureloop).
You shall not change display content while in the picture-loop! - Or you have to redraw the entire screen.

lovethat777, 2019-08-05:
This is clearly a bug that needs to be addressed. I'm seeing it consistently.

boelle, 2019-08-06:
#define LCD_LANGUAGE en
#define DISPLAY_CHARSET_HD44780 JAPANESE

i always change it to WESTERN as i dont read or speak japanese

i never looked in to what the difference is but i have never had an issue when set to western

7eggert, 2019-08-06:
The DISPLAY_CHARSET_HD44780 is the burned-in character-set of your device. As long as you're using A-Z only, you won't notice an error, but for European languages, you'll see wrong characters on your **non-graphical** screen (not the screen used by the bug reporter).

lovethat777, 2019-08-06:
Indeed, DISPLAY_CHARSET_HD44780 is the hardcoded charset of the device and cannot be changed by simply re-defing that directive.

> In the upper half of the screen is still half of a "1" visible.
> This is not a noise problem but a consequence of violating u8gLibs [picture [...]

This seems to be the problem

boelle, 2019-09-24:
@lovethat777 is the issue still there?

xyyx, 2019-09-29:
I have same issue on skr 1.3 and biqu reprap display clone

boelle, 2019-10-21:
i have put the confirmed bug label on this one,  using a re-arm myself i have never seen this, but then again i use english as language (thou i'm from DK)

boelle, 2019-11-09:
So how do we fix this one for good?
@AnHardt @thinkyhead @shitcreek

boelle, 2019-12-02:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-12-02 by boelle