MarlinFirmware/Marlin #14836: [BUG] (short description)

jimmyeao, 2019-08-04:
### Description

Babysetting menu missing from Tune Menu

### Steps to Reproduce

```
#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
  #define BABYSTEP_WITHOUT_HOMING
  //#define BABYSTEP_XY                     // Also enable X/Y Babystepping. Not supported on DELTA!
  #define BABYSTEP_INVERT_Z false           // Change if Z babysteps should go the other way
  #define BABYSTEP_MULTIPLICATOR  1         // Babysteps are very small. Increase for faster motion.

  //#define DOUBLECLICK_FOR_Z_BABYSTEPPING  // Double-click on the Status Screen for Z Babystepping.
  #if ENABLED(DOUBLECLICK_FOR_Z_BABYSTEPPING)
    #define DOUBLECLICK_MAX_INTERVAL 1250   // Maximum interval between clicks, in milliseconds.
                                            // Note: Extra time may be added to mitigate controller latency.
    #define BABYSTEP_ALWAYS_AVAILABLE     // Allow babystepping at all times (not just during movement).
    //#define MOVE_Z_WHEN_IDLE              // Jump to the move Z menu on doubleclick when printer is idle.
    #if ENABLED(MOVE_Z_WHEN_IDLE)
      #define MOVE_Z_IDLE_MULTIPLICATOR 1   // Multiply 1mm by this factor for the move step size.
    #endif
  #endif

  #define BABYSTEP_DISPLAY_TOTAL          // Display total babysteps since last G28

  //#define BABYSTEP_ZPROBE_OFFSET          // Combine M851 Z and Babystepping
  #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
    //#define BABYSTEP_HOTEND_Z_OFFSET      // For multiple hotends, babystep relative Z offsets
    //#define BABYSTEP_ZPROBE_GFX_OVERLAY   // Enable graphical overlay on Z-offset editor
  #endif
#endif
```
**Expected behavior:** Able to change z offset when printing

**Actual behavior:** Option not available on LCD under Tune

#### Additional Information
Board, Bigtreetech SKR 1.3 with TMC 2208 Drivers (Legacy mode)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3464854/Configuration.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

firepower9966, 2019-08-04:
babysteps is enabled by double clicking and also in tune menu when print started on mine using marlin from 3/7/2019 on my mega2560

jimmyeao, 2019-08-04:
> babysteps is enabled by double clicking and also in tune menu when print started on mine using marlin 2.0 bugfix from 3/7/2019 on my mega2560

On the SKR 1.3 with Marlin Bugfix from 27th July babystepping does not appear in the Tune menu. Double clicking also does not work

InsanityAutomation, 2019-08-04:
Double clicking doesnt work because its disabled

//#define DOUBLECLICK_FOR_Z_BABYSTEPPING

Code definitely shows it in the tune menu. Just verified the current snapshot.

jimmyeao, 2019-08-04:
> Double clicking doesnt work because its disabled
> 
> //#define DOUBLECLICK_FOR_Z_BABYSTEPPING
> 
> Code definitely shows it in the tune menu. Just verified the current snapshot.

It is now, even with it uncommented, it does nothing

jimmyeao, 2019-08-04:
Resolved. Octoprint wasn't flashing the board correctly (despite reporting success). Manually flashed and working as expected

closed 2019-08-04 by jimmyeao