MarlinFirmware/Marlin #14837: Marlin 2.0 MKS Sbase Missing Pin Macro?

MicrowavedBread, 2019-08-04:
### Description
While trying to save the program it gives the errors:
Missing binary operator before token "(" on lines 137:12, 161:12, 266:15, and 289:15
http://prntscr.com/oo4ws4
It happens wit the latest release and the one in the zip file I attached.
I read in another post that the macro "LPC1768_PIN_INTERRUPT" was accidentally removed and i guess not put back in since october.
https://github.com/MarlinFirmware/Marlin/issues/12166

### Steps to Reproduce

1. Change to MKS Sbase board
2. Try to save or compile

**Expected behavior:** [What you expect to happen]
No errors and a working probe

**Actual behavior:** [What actually happens]
Errors and no working probe

p3p, 2019-08-04:
You need to supply your configs and probably the pins file you are using, you forgot to attach the zip file you mention. 

The issue you linked was fixed long ago, and all our tests are passing so its something you are changing for your setup.

MicrowavedBread, 2019-08-04:
[Config and pin files.zip](https://github.com/MarlinFirmware/Marlin/files/3465393/Config.and.pin.files.zip)

Sorry, i must have accidentally deleted it

p3p, 2019-08-04:
All 3 of these files are very out of date, you will need to transfer your settings from them into the current versions.

This was the first error when I tried to build:
`Marlin/src/HAL/HAL_LPC1768/../../core/../inc/SanityCheck.h:374:4: error: #error "LPC_SD_(LCD|ONBOARD|CUSTOM_CABLE) are now SDCARD_CONNECTION. Please update your Configuration_adv.h."`

After seeing this I did a diff between your files and the current ones and it showed considerable changes not related to your settings.

MicrowavedBread, 2019-08-04:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3465432/Marlin.zip)
I'm getting the same issue on the latest version from github

thinkyhead, 2019-08-04:
> I read in another post that the macro "LPC1768_PIN_INTERRUPT" was accidentally removed and i guess not put back in since October.

According to my reading of that issue, `LPC1768_PIN_INTERRUPT_M` is now part of the framework so we don't need to define it.

p3p, 2019-08-04:
What IDE are you using and how long has it been since you updated your libraries, frameworks / platforms? You need to keep them up to date. (although I haven't changed anything in so long I almost forgot to mention it)

MicrowavedBread, 2019-08-04:
I'm using Platformio in Atom. And everything is up to date as far as i can tell

thinkyhead, 2019-08-04:
Please use Visual Studio Code. PlatformIO is no longer supporting Atom.

p3p, 2019-08-04:
If you're up to date, when you click `platformio->update platforms packages and libraries`  it should show 
```
Platform NXP Arduino LPC176x
--------
Updating nxplpc-arduino-lpc176x          @ 0.0.1          [Detached]
Updating toolchain-gccarmnoneeabi        @ 1.80201.181220 [Up-to-date]
Updating tool-pyocd                      @ 0.801.0        [Up-to-date]
Updating framework-arduino-lpc176x       @ 0.1.3          [Up-to-date]
```

Also what thinkyhead said.

thinkyhead, 2019-08-04:
Successfully built MKS SBASE test configurations. No issues.

closed 2019-08-05 by thinkyhead

MicrowavedBread, 2019-08-05:
Im getting the same issue in visual studio code
http://prntscr.com/ooc4oq

MicrowavedBread, 2019-08-05:
I deleted both of my modified marlin folders and grabbed a completely untouched fresh one and it still gives the same errors

p3p, 2019-08-05:
can you test with the minimum changes, so a fresh download and just set the motherboard.

MicrowavedBread, 2019-08-05:
> 
> 
> can you test with the minimum changes, so a fresh download and just set the motherboard.

For some reason the mks sbase board isnt on the one i just downloaded. I don't think im that blind