MarlinFirmware/Marlin #14840: AGCM4 linker issue

GMagician, 2019-08-04:
I'm working with vscode + pio extension and when compiling for debugging I get this compile error:

> .pioenvs/adafruit_grandcentral_m4/src/src/module/planner.cpp.o: In function `Temperature::scaledFanSpeed(unsigned char, unsigned char)':
c:\Users\Giuliano\Documents\GitHub\Marlin/Marlin\src\module/temperature.h:484: undefined reference to `Temperature::fan_speed_scaler'

this error is raised when -Os optimization is replaced by -Og (automatically done by pio extension)

and cause is `constexpr`.

I'm wondering, no one else has this issue trying to debug code?
How can be changed code to prevent this? I think that removing `constexpr` is not a "permitted" solution then?

p3p, 2019-08-04:
What c++ standard are you building with? I don't have any issues building `-std=gnu++17 -Og`, and if it builds with other optimisation levels something odd is going on.

thinkyhead, 2019-08-05:
```
$ mftest
  1) DUE
  2) LPC1768
  3) LPC1769
  4) STM32F103R
  5) STM32F103R_alfawise
  6) STM32F407VE_black
  7) adafruit_grandcentral_m4
  8) esp32
  9) linux_native
 10) megaatmega2560
 11) teensy31
 12) teensy35

Select a test to apply (1-12) : 7

Available 'adafruit_grandcentral_m4' tests:
  1) Build Grand Central M4 Default Configuration

#
# Test adafruit_grandcentral_m4 (1) 
#
restore_configs
opt_set MOTHERBOARD BOARD_AGCM4_RURAMPS4D_13

Build adafruit_grandcentral_m4 test #1 (y/N) ? y
Processing adafruit_grandcentral_m4 (platform: atmelsam; board: adafruit_grandcentral_m4; framework: arduino)
. . .
Environment adafruit_grandcentral_m4	[SUCCESS]
```

GMagician, 2019-08-05:
> What c++ standard 

It's Marlin default, c++12 I think, tool chain was old (when I checked) and didn't know c++17

@thinkyhead error is when running 'pio debug' via vscode. Don't know who add such options but redirecting output debug mode add -Og and -ggdb3. -Og is the problem (it you add -Og in platformio.ini you will see what I means (servo must be enabled to see it, bltouch option for example)

GMagician, 2019-08-05:
fixed by #14838 just addinf gnu++17 as suggested by @p3p. A lot of warning for deprecated "register" will appear but that is inside framework

closed 2019-08-05 by GMagician