MarlinFirmware/Marlin #14843: IDEX not working with TMC 2130's installed on X axis's with sensorless homing

hallphilip, 2019-08-05:
IDEX not working with TMC 2130's installed on X axis's. I have this feature all working as long as I install drv8825's on X with manual endstops. When I try and go back to my TMC2130 with sensorless home the X2 driver will not enable. The driver is confirmed to be working as well as the driver slot. I have not yet tried with the TMC 2130 with sensorless home disabled. I will try that tomorrow. Is this just not possible yet?

Roxy-3D, 2019-08-05:
I don't believe TMC2130's don't work with IDEX.  The IDEX code is many layers above the IDEX code.  The IDEX code doesn't know what kind of stepper motor drivers you have in your system.

My guess is you don't have the 2nd X stepper motor pins defined correctly.

InsanityAutomation, 2019-08-05:
I will say it will be troublesome for idex for another reason however. Repeatability with sensorless homing is not likely to net good results aligning the hotend X positions on dual material prints.

Roxy-3D, 2019-08-05:
Oh Geeeze!  I didn't think of that corner case!   Yeah, I bet there are alignment issues.   But then again, if some special code was written to do the sensor-less homing very accurately...   maybe that can be overcome.

hallphilip, 2019-08-07:
Thanks for the input guys, I have started installing dual endstops. I was sure my pins were set correctly but it never hurts to have a second look. Everything else seems to work fine for me otherwise.

boelle, 2019-09-24:
@hallphilip is the problem still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle