MarlinFirmware/Marlin #14847: [FR] use platformio `extra_configs` attribute

trouch, 2019-08-05:
I think it would be nice to move PIO envs to another location using `extra_configs` attributes. platformio.ini is slowly becoming a monster, that could help maintaining and adding envs in the future 😅

```
[platformio]
src_dir       = Marlin
build_dir     = .pioenvs
lib_dir       = .piolib
libdeps_dir   = .piolibdeps
boards_dir    = buildroot/share/PlatformIO/boards
default_envs  = megaatmega2560
extra_configs  = buildroot/share/PlatformIO/envs/*.ini
```

**Benefits:**
* We could have as much ini file we'd like, eventually per env/family/mcu, similar to pins file maybe.
* Main platformio.ini will be used for `[platformio]`, `[common]` and reference envs (eg. mega, linux, ...). No more modification when adding a new env for a particular board.
* We could even have a user folder

I wanted to submit a PR, but thought to ask first how you'd like to organize environments.

thisiskeithb, 2019-08-25:
> I wanted to submit a PR, but thought to ask first how you'd like to organize environments.

Probably by HAL, like [what was done for the pins](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin/src/pins).

tpruvot, 2019-08-25:
if it loads all ini files from a folder, could be one file per HAL indeed, but for vendor forks, it can be whatever the vendor want... like its vendor name... unsure multiple *.ini for several folders are supported, nor a good idea... imo buildroot/share/PlatformIO/envs/ is too deep/hidden for users...

isn't windows explorer using some system ini files, i remember some "desktop.ini"

trouch, 2019-08-26:
Yeah could be anything 😅 
I'd rather wait for @thinkyhead suggestion how he'd like it.

@tpruvot indeed .ini file was a Windows format, used widely now. Still used for folder customization with the _hidden_ `desktop.ini` file as you mentioned.

To what I tested, `extra_configs` takes any number of entries with proper wildcard support.

That's something I already use in my own branch to define several builds with shared Marlin configuration files.

slowbro, 2021-07-03:
Pretty sure this was done, since all the envs are in `ini/` now.

thisiskeithb, 2021-07-03:
Resolved in https://github.com/MarlinFirmware/Marlin/pull/21507.

closed 2021-07-03 by thisiskeithb