MarlinFirmware/Marlin #14849: [BUG] compile error with auto chamber fan

maukcc, 2019-08-05:
When compiling with auto chamber fan, we get this error:

> /tmp/arduino_build_125417/sketch/src/module/temperature.cpp: In static member function 'static void Temperature::checkExtruderAutoFans()':
> /tmp/arduino_build_125417/sketch/src/module/temperature.cpp:707:13: error: 'chamberfan_speed' was not declared in this scope
>              chamberfan_speed = fan_on ? CHAMBER_AUTO_FAN_SPEED : 0;

shitcreek, 2019-08-06:
I can't reproduce this error. By default CHAMBER_AUTO_FAN_SPEED is declared in configuration_adv.h, please check to see that it is.

maukcc, 2019-08-06:
this is in bugfix 2.0.

> #define CHAMBER_AUTO_FAN_SPEED        255

tried also a couple of other values, with the same result

did you try with a pin defined

> #define CHAMBER_AUTO_FAN_PIN 10

and a temp sensor defined

> #define TEMP_SENSOR_CHAMBER 1

otherwise this is ignored.

shitcreek, 2019-08-06:
Gotcha, here's the issue: PSU_CONTROL needs to be enabled since AUTO_POWER_CONTROL is nested inside of its conditional clause.

maukcc, 2019-08-06:
Thanks !
I think this needs some cleanup, as there are some machines out there that have no PSU control.
As a workaround we copied the #define AUTO_POWER_CHAMBER_FAN outside of the PSU control, as enabling PSU control gave us a lot of errors.

Also:
At this point the fan goes on when the heat goes over the specified temp, an off when it gets below.
Can it be made so that the fan goes on when the heat is set above the temp, and of when it is set below the temp.
The point is that we set the CHAMBER_AUTO_FAN_TEMPERATURE  at  28, now when the command comes to heat to 65, the fan will not go on until it temps reaches 28. by the time the sensor(which is mounted half way the height) reads 28, the top of the chamber is already at 50 because the air was not spread around by the fan.

shitcreek, 2019-08-06:
You would need to move this entire segment outside of it:
`#define AUTO_POWER_CONTROL  // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS           // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    #define AUTO_POWER_CONTROLLERFAN
    #define AUTO_POWER_CHAMBER_FAN
    //#define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU over this temperature
    #define POWER_TIMEOUT 30
  #endif`

The reason your heater is coming on is because typically pin 10 is assigned to the heater. Use a different pin that isn't in use.

I'll look into that last bit tomorrow.

maukcc, 2019-08-06:
Also: 
the chamberheat does not seem to trigger the TEMP_STAT_LEDS

maukcc, 2019-08-09:
And probably related:
When we do M106 Sxxx or M107 we get:
Recv: echo:Unknown command: "M106 S255"

timmmoore, 2019-08-09:
When the recent PSU_CONTROL changes were done, it was clear the auto_fan stuff under PSU_CONTROL was when no pin controlled the fan and the auto fan stuff in configuration_adv.h were when a fan was controlled by a pin. I think you have a chamber fan controlled by a pin? so the PSU_CONTROL stuff doesn't matter.
The comment that M106/M107 doesn't work says it thinks there are no fans. Looking in conditionals_post.h there is no check for chamber fan for fan_count.
I assume you have no fans except the chamber, so fan_count is 0. that explains the M106 issue.
There looks to be a 2nd issue when you define a temp chamber sensor/adc and fan pin in configuration_adv.h the code in temperature.cpp comes in but the variable holding the chamber fan speed doesn't so you get a compiler error. Looks likes
    #if ENABLED(AUTO_POWER_CHAMBER_FAN)
      static uint8_t chamberfan_speed;
    #endif
in temperature.h and cpp needs changing to allow for case of auto chamber fan via a pin.
e.g. changing the above to
    #if ENABLED(AUTO_POWER_CHAMBER_FAN) || HAS_AUTO_CHAMBER_FAN
      static uint8_t chamberfan_speed;
    #endif
compiles. The M106 is a differnt problem.

timmmoore, 2019-08-09:
Looking at M106/107 it only supports part and extruder fans. Controller and chamber fans are not supported with the current code.

closed 2019-09-23 by maukcc