MarlinFirmware/Marlin #14850: Missing temperature icon on 1602 LCD

Eszartek, 2019-08-05:
While  compiling Marlin for a new SKR 1.3 board, I found that the resulting firmware would display a 'y' in place of the thermometer icon (char) in the upper left of my 1602 LCD. After playing with all the configuation.h LCD options, I realized that the temperature icon would render properly if I just chose the REPRAP_DISCOUNT_SMART_CONTROLLER , but since it is typically a 20x4 display, the result was that the menus were not usable.

After looking at the source, I found that in  ultralcd_HD44780.cpp MarlinUI::draw_status_screen() , the drawing of the status for screens < 20
used -1 instead of LCD_STR_THERMOMETER[0]. 

So the simple fix was to change line 753 from:
```  _draw_heater_status(H_E0, -1, blink);```

to:

```  _draw_heater_status(H_E0, LCD_STR_THERMOMETER[0], blink);```

Now the LCD renders the temperature icon as it should. 

The last bit to get the menus working was to just add:
```
  #define LCD_WIDTH 16
  #define LCD_HEIGHT 2
```
below the uncommented define for the REPRAP_DISCOUNT_SMART_CONTROLLER.

shitcreek, 2019-08-06:
question: are you using LCM1602 generic LCD adapter?

Eszartek, 2019-08-06:
No LCM1602 (i2c) adapter, the interface from my display/encoder board is the two 10pin exp1/exp2 interface cables like the larger 20x4 display/encoder boards. Im not sure what the original make model of my display board is, as I havent removed it from the printer.

shitcreek, 2019-08-06:
I've added a CUSTOM_LCD_SIZE condition that would allow users to change the size and your fix.

boelle, 2019-08-06:
could we considered this as fixed or should we wait for the merge?

shitcreek, 2019-08-06:
We could be waiting a long time, haha, I am impartial.

thisiskeithb, 2019-08-06:
> could we considered this as fixed or should we wait for the merge? 

Waiting until the fix is actually implemented would be ideal.

shitcreek, 2019-08-07:
> We already auto-detect the width and height of the LCD based on model. Instead of adding a custom option, simply get the configurations from the original bug report and see which LCD selection has the problem, then add the appropriate conditions to `Conditionals_LCD.h` to assign the correct width and height for the given model.
- @thinkyhead

thinkyhead, 2019-08-07:
The `REPRAP_DISCOUNT_SMART_CONTROLLER` is a 20x4 LCD. If that's the option you've been forced to use by your wiring, then of course you can override the width and height for your purposes. But this is not something we need to support widely.

closed 2019-08-07 by thinkyhead