MarlinFirmware/Marlin #14854: UBUNTU

thecraftman, 2019-08-05:
I have an issue running this o my ubuntu terminal

shitcreek, 2019-08-06:
To get help, please provide more information than just this such as config files, etc.

That said:
We're not a support group
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

reloxx13, 2019-08-06:
r u serious opening an issue with zero informations?!

boelle, 2019-08-06:
yeah, and as a dev person he should know better

not to sound agressive i say we should close, but who agree?

shitcreek, 2019-08-06:
I'd give him a day or two to respond. I'm guessing baud rate is off.

boelle, 2019-08-06:
oh yes, they must be set to the same :-)

thecraftman, 2019-08-07:
Thanks all tho . i got it fixed is all good now ..

closed 2019-08-07 by shitcreek