MarlinFirmware/Marlin #14865: [FR] SKR 1.3 + TMC2130 + delta SENSORLESS_PROBING and SENSORLESS_HOMING

syproduction, 2019-08-06:
### Description

I observe Marlin is not fully compatible to Deltas+TMC2130+SKR 1.3+SENSORLESS
First, in Marlin Deltas should home to MAX Endstops but on SKR board DIAG1 pin on stepStick hardwired to MIN Endstops. In this case I had to swap pins in pins_BIGTREE_SKR.h, but that feels just not right.

Then, Z probing cannot use the same pin as Endstop. I chose Z axis endstop with Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and then connected Z_MIN_PIN to Z_MAX_PIN with wire on board. Again, it feels just not right.

And then, why should i choose Z axis for probing on Delta? Should it be something integral on all axis for SENSORLESS_PROBING?

Also when homing/probing it is desired to change some values temporarily with GCODE. For example we need to hit ends fast, just one time, with TMC driver current, stallGuard sensivity and Feedrate different from normal operation. And then put values back to normal.

### Request
1. Permit homing to MIN Endstops on SKR Board 1.3 when SENSORLESS_PROBING and SENSORLESS_HOMING are active
2. Permit using same pin as Z probe and Z endstop or make a feature when any endstop hit is understood as Z probing when SENSORLESS_PROBING is active.
3. Allow custom GCODE routines for homing and probing processes.

mur-lin, 2019-08-07:
This what I use in pins file:

// The TMC2130 DIAG1 pins are board jumper connected to MIN_PIN's
// Delta sensorless homing need these connected to MAX_PIN's
//
#if BOTH(DELTA, SENSORLESS_HOMING)
  #define X_MIN_PIN          P1_28
  #define X_MAX_PIN          P1_29
  #define Y_MIN_PIN          P1_26
  #define Y_MAX_PIN          P1_27
  #define Z_MIN_PIN          P1_24
  #define Z_MAX_PIN          P1_25
#else
  #define X_MIN_PIN          P1_29
  #define X_MAX_PIN          P1_28
  #define Y_MIN_PIN          P1_27
  #define Y_MAX_PIN          P1_26
  #define Z_MIN_PIN          P1_25
  #define Z_MAX_PIN          P1_24
#endif

McVillano, 2019-08-12:
+1 to the petition, we need more delta suppor please!

McVillano, 2019-08-12:
@mur-lin @syproduction you are my heros! I was strungling with this with no clue and with this workarround i was able to make it work!
Thanks!

mylife4aiurr, 2019-08-12:
+1M for more delta love

McVillano, 2019-08-18:
@mylife4aiurr @mur-lin @syproduction have youb guys managed to make a delta autocalibration? I have an anycubic kossel plus and allways in interation 2 or 3 fails.

mylife4aiurr, 2019-08-18:
@McVillano Yes G33 working for me in bugfix 2.0x , no issues.

McVillano, 2019-08-18:
@mylife4aiurr with sensorless homming? What are you values of Z_Homming_Feedrate and sensibility on the towers??

mylife4aiurr, 2019-08-18:
Misunderstood your question. Thought u wanted to know if any 1 got g33 working:
"have youb guys managed to make a delta autocalibration?"

so yes to g33, no I didnt do sensorless bed probing with tmc2130's

McVillano, 2019-08-19:
> 
> 
> Misunderstood your question. Thought u wanted to know if any 1 got g33 working:
> "have youb guys managed to make a delta autocalibration?"
> 
> so yes to g33, no I didnt do sensorless bed probing with tmc2130's

Not sensorless bed probing, just sensorless homming, i'm trying to make a G33 with TMC2130 as endstop and i've not managed to do it, always fails in second or third iteration

syproduction, 2019-08-22:
Hello. After spending too much time on trying to make everything work... I abandoned it. 
Actually SENSORLESS function looks not so useful for delta. 
But StallGuard error rehoming is not implemented at all. We can see some implementation in Prusa's Marlin firmware on their website but none in official Marlin.
Actually if we somehow enable pause and rehome on errors, StallGuard can work just like regular endstop.

Murray-Lindeblom, 2019-08-22:
On my Delta, I have found the sensitivity required depended on the starting position of the effector.  Thinking the GT2 belt was a significant factor, I switched to a Chinese Gates clone belt.
 The Gates clone belt is definitely better; however: I an questioning it I need to restore my original optical end stops.

Allarabie, 2019-12-16:
I have the exact same setup (Deltas+TMC2130+SKR 1.3+SENSORLESS) and I finally found something that worked for me.!!!

I have been struggling with this for a few days now, and finally found the culprit for my issue.
#define MINIMUM_STEPPER_PULSE 2 is incorrect!.
2 is NOT the correct pulse value for TMC2130s, 0 is..
I comment this line out all together and used default value.
This is the one thing that actually made sensorless homing trigger, at all. 
Until I commented this out, nothing was making a difference and StallGuard was never triggering, the carriages were just smashing into the top of the towers till the printer killed itself.

So Yes, you do need to have all the correct settings enabled to begin with, but even once you do, this issue will probably still be a problem.  At least it was for me.

Also, separate issue (and FYI for Devs maybe), but very important...
The homing code for Delta is not correct for Sensorless homing. 
See Thinkyhead's comment from Mar 17 2018. https://github.com/MarlinFirmware/Marlin/issues/10141
This is still incorrect in the latest Marlin 2.0. However it is now located in the delta.ccp file.
Homing will not work properly until you fix the code as Thinkyhead suggests.

So. I now have Sensorless Homing working.  
Now onto Sensorless Probing.
So far, the nozzle moves down to the bed, presses down a bit, then I get:
READ: Error:Probing Failed
READ: Correct delta settings with M665 and M666
So, Im still working on that part. ;-)

Hope this helps someone.

Hywelmartin, 2019-12-17:
In Klipper it's supposed to work... BUT only with the function [Stepper phase adjusted endstops](https://github.com/KevinOConnor/klipper/blob/9c25f42f4a35f4e1ad08e730e25a0094dd2986e2/config/example-extras.cfg#L354) and endstop_align_zero: True

https://www.facebook.com/groups/anycubic3dp/permalink/1001673216861556/
[image]

mylife4aiurr, 2020-02-03:
Willing to help with this------>Delta with sensorless homing and sensorless probing.
Just downloaded Marlin 2.0.3: 
Have:
skr v1.3
skr v1.4
tmc 2130 official, 2208 bigtreetech, 2209 v1.2 bigtreetech
bltouch
(current hardware skrv1.3, 2208, hardware endstops)

Tried to complied for Marlin 2.0.3 skr1.4, tmc 2209, senorless homing, senorless probing:
```
Marlin\src\module\probe.cpp: In static member function 'static float Probe::run_z_probe()':
Marlin\src\module\probe.cpp:592:18: error: 'TOTAL_PROBING' was not declared in this scope
     float probes[TOTAL_PROBING];
                  ^~~~~~~~~~~~~
Compiling .pio\build\LPC1768\src\src\module\stepper\TMC26X.cpp.o
Marlin\src\module\probe.cpp:592:18: note: suggested alternative: 'EXTRA_PROBING'
     float probes[TOTAL_PROBING];
                  ^~~~~~~~~~~~~
                  EXTRA_PROBING
Marlin\src\module\probe.cpp:623:34: error: 'p' was not declared in this scope
         for (uint8_t i = 0; i <= p; i++) {                            // Iterate the saved Zs to insert the new Z
                                  ^
Marlin\src\module\probe.cpp:624:25: error: 'probes' was not declared in this scope
           if (i == p || probes[i] > z) {                              // Last index or new Z is smaller than this Z
                         ^~~~~~
Marlin\src\module\probe.cpp:624:25: note: suggested alternative: 'probe'
           if (i == p || probes[i] > z) {                              // Last index or new Z is smaller than this Z
                         ^~~~~~
                         probe
*** [.pio\build\LPC1768\src\src\module\probe.cpp.o] Error 1
```

ellensp, 2020-02-04:
Use bugfix-2.0.x It compiles fine with skr1.3/skr1.4, tmc 2209, sensorless homing and sensorless probing. but I can't test it...  
Here are my only slightly modified from example delta configurations. 
[delta-sensorless-skr14.zip](https://github.com/MarlinFirmware/Marlin/files/4150868/delta-sensorless-skr14.zip)

ellensp, 2020-02-04:
"Permit homing to MIN Endstops on SKR Board 1.3 when SENSORLESS_PROBING and SENSORLESS_HOMING are active"
you can compile with home MIN or home MAX on SKR Board 1.3 when SENSORLESS_PROBING and SENSORLESS_HOMING without and issues.

labotecno, 2020-06-03:
hello i try to get it with marlin 2 last version , when i try to put probe sensor less endstop max dont working!  my tmc2130 is ready to use with skr11.3 ... someone can to help me to get this feature

Kadah, 2020-08-11:
Unless I'm missing something, sensorless probing on a delta doesn't seem like it should be reliable for any point other 0,0.

closed 2022-01-20 by thisiskeithb