MarlinFirmware/Marlin #14870: [BUG] lib_ignore does not seem to completely work with platformio aliases

TheZoker, 2019-08-07:
### Description

It seems that the platformio alias ignore from this commit does not fully work:
https://github.com/MarlinFirmware/Marlin/commit/ab792d004cf22722088e0021c34f84724daa32ca

You can see it on this branch:
https://github.com/TheZoker/Marlin/commits/skr-e3-dip

With the default `lib_ignore` for the `BIGTREE_SKR_MINI` board, the build will fail:
https://github.com/TheZoker/Marlin/commit/33f33234df4398a60d01d3f33b59882f19ea3911#diff-237285c047dad808f73698c139144e1a

When I re-add `c1921b4` to the `libs_ignore` (like it was before), it builds:
https://github.com/TheZoker/Marlin/commit/39f5e23cf72e5d2f253bb67c99308e7f6b825548

Here is the line, that causing the issue:
https://github.com/MarlinFirmware/Marlin/commit/ab792d004cf22722088e0021c34f84724daa32ca?diff=split#diff-237285c047dad808f73698c139144e1aL323 

This is the log, when the build fails:
https://gist.github.com/TheZoker/5201c4bf9caf2cb05bdb65993e04cb03

### Steps to Reproduce

1. Try to build Marlin with [this](https://github.com/TheZoker/Marlin/commit/33f33234df4398a60d01d3f33b59882f19ea3911#diff-237285c047dad808f73698c139144e1a) commit 
2. Will not build

**Expected behavior:** build
**Actual behavior:** not building

Bansh1990, 2019-08-08:
"@tpruvot tpruvot force-pushed the tpruvot:stm32f1_tmc_serial branch from 943a543 to b76bf93 2 days ago"

after this dont compile on skr e3 dip
and compiling after delete this files

thinkyhead, 2019-08-19:
Some changes have been made to lib_ignore throughout platformio.ini in the latest code, so give it a try and see how it fares. You may want to delete the platformio work folders before doing any builds.

boelle, 2019-09-24:
@TheZoker tried what thinky suggested?

TheZoker, 2019-09-24:
Sorry forgot to answer. Yes it worked, thank you!

closed 2019-09-24 by TheZoker