MarlinFirmware/Marlin #14872: [BUG] SKR mini V1.1 can not compile with branch 4378

Udo-Nbg, 2019-08-07:
Could not Compile branch 4378 with SKR Mini V1.1

The previous branch 4350 i can compile and most Featutes (not fully integrated yet) works fine (TXT/GFX Display, Hotend, Thermistors, Fan)

The Setings are identical.
also when i copy the "configuration.h" from branch 4350 to branch 4378 i get the same errors.

Ludy87, 2019-08-07:
what is the error message?

Udo-Nbg, 2019-08-07:
Her are the complet Compiling Logs...
[Error Compiling-LOG SKR-MINI_V1.1 Marlin-bugfix-2.0.x branch 4378.txt](https://github.com/MarlinFirmware/Marlin/files/3478317/Error.Compiling-LOG.SKR-MINI_V1.1.Marlin-bugfix-2.0.x.branch.4378.txt)
[Success Compiling-LOG SKR-MINI_V1.1 Marlin-bugfix-2.0.x branch 4350.txt](https://github.com/MarlinFirmware/Marlin/files/3478319/Success.Compiling-LOG.SKR-MINI_V1.1.Marlin-bugfix-2.0.x.branch.4350.txt)

Ludy87, 2019-08-07:
The bugfix is still in the waiting loop
https://github.com/MarlinFirmware/Marlin/pull/14861

Ludy87, 2019-08-08:
@Udo-Nbg
fix is deployed

Udo-Nbg, 2019-08-08:
branch 4382
it compile successful
and it should work

Thank You

closed 2019-08-08 by Udo-Nbg

Bansh1990, 2019-08-08:
`thinkyhead added some commits 2 hours ago
 @thinkyhead
Identify SoftwareSerial.h as a placeholder
b848fdb
 @thinkyhead
Update SoftwareSerial.h
ca6dfab
`
after this dont compile on skr e3 dip