MarlinFirmware/Marlin #14874: [BUG 2.0.x] GCODE Baudrate always "implausible"

reloxx13, 2019-08-07:
### Description

I wanted to try a higher baudrate.   
For this i enabled `BAUD_RATE_GCODE` in the config_adv.   

My default baudrate is set to 250000.   

Command send with Octoprint Terminal.

### Steps to Reproduce
- Enable `BAUD_RATE_GCODE` in config_adv   
- Change the Baudrate by GCODE with M575 S1000000

**Actual behavior:**
The Printer does not change the baudrate and responses with:
```
Send: M575 S1000000
Recv: echo:?(B)aud rate is implausible.
Recv: ok
```

```
Send: M575 S2400
Recv: echo:?(B)aud rate is implausible.
Recv: ok
```

This happens with all suggested Baudrates   
`[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]`

#### Additional Information
Gcode doc: http://marlinfw.org/docs/gcode/M575.html
Config: https://github.com/reloxx13/Marlin/tree/my-skr-mini-e3/Marlin
Ender3 with SKR mini E3 board https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-
Latest OctoPrint on Pi3b+

thisiskeithb, 2019-08-07:
`M575` support is still buggy, but @InsanityAutomation is working on it: https://github.com/MarlinFirmware/Marlin/pull/14793.

reloxx13, 2019-08-07:
Okay thanks, did not see that PR

closed 2019-08-07 by reloxx13

G2Barbour, 2020-06-08:
I just had this issue and upon looking at the code, it appears that the documentation is what is wrong. 'S' is not the parsed parameter for baud, 'B' is. That does not mean it does not have other problems. I have not tested it fully, but was able to successfully change values more then once.

thinkyhead, 2020-08-07:
Documentation has been fixed.