MarlinFirmware/Marlin #14878: [FR] adaptive current on TMC2209

Gerggg, 2019-08-08:
Hello,
For the Marlin 2.0, on the TMC2209 in UART mode, is the adaptive current function implemented?
Thank you

teemuatlut, 2019-08-08:
No. Hard to tune, understand tuning and verify it's actually working.
Just setting the bits isn't enough to pass for a feature.

Gerggg, 2019-08-19:
I understand the difficulty of implementing this kind of function but mastering the heating of the motors ensuring that you always have enough power for optimal operation will be really cool.

thisiskeithb, 2024-03-09:
There was an unsuccessful effort to implement this and in the end, it was reverted.

See [these PRs](https://github.com/MarlinFirmware/Marlin/pulls?q=is%3Apr+coolstep+is%3Aclosed+in%3Atitle) for more information.

closed 2024-03-09 by thisiskeithb