MarlinFirmware/Marlin #14880: [BUG] SKR mini 1.1 + TMC2130 no movement

IvanLev, 2019-08-08:
### Description

When installing TMC2130 on SKR mini 1.1 board with stm32f1 MCU the stepper drivers don't work. They completely ignore the en signal and don't power up the stepper motors.

M122 doesn't show any errors and can read the registers correctly.
[image]
I suspect that the issue is in msteps as they should be configured as 16 in my config file.
Strange that the currents are written correctly to the drivers but msteps not. In my configuration_adv.h file they are set to 16.

#### Config files
[Config_tmc.zip](https://github.com/MarlinFirmware/Marlin/files/3482002/Config_tmc.zip)

IvanLev, 2019-08-11:
Setting the drivers into spi mode lets the stepper move. This is with the firmware for spi mode. So it confirms my theory with msteps. I am ready to debug it myself if someone can help me start debugging in vscode. Tried flashing the firmware using st-link but it didn't work.

tpruvot, 2019-08-12:
for the stlink you need to comment the .py relocation script (but if you flashed via stlink has erased the optional vendor boot loader, which is not required to boot, just different)

boelle, 2019-09-24:
@IvanLev still having problems?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle