MarlinFirmware/Marlin #14884: [BUG] E steps settings only in hundreds

kihope, 2019-08-08:
When i want set esteps in LCD only values are 0 - 100 - 200 - 300 etc, setting it using configuration.h doesnt work neither, my board is skr 1.3

macasero, 2019-08-08:
same here, also Initialitate eprom, doesn´t seam to work fine.

Pronterface M92 setting, M500 and M503 shows correct steps, but lcd settings only show units and set to hundreds

shitcreek, 2019-08-08:
What version of Marlin are y'all using? Please provide more information.

kihope, 2019-08-08:
I have Marlin 2, i downloaded about day ago when i was updating the printer

shitcreek, 2019-08-08:
Could you provide your configs? What happens when you turn the dial fast vs slow?

kihope, 2019-08-08:
https://github.com/kihope/TevoTrantula_Marlin-2.0_SKR_v1_3/blob/master/Configuration.h

Dont mind the mess please, i can change steps in config, however when i go to lcd all goes by hundred, so i even cant get the values i had before back, before step were able se by tenths if i remember now only by hundreds (as i mentioned: 0,100,200,nnn.. way up) i dont think it is caused by my config..

shitcreek, 2019-08-08:
I don't see anything out of place in your config.
Are you using JimBrown's easyConfig 2.0 or Marlin bugfix 2.0?

tpruvot, 2019-08-09:
see e7b3720b8193c76d1766898923b59261371e1c79 for a possible way to fix that. btw the default config for E steps are incredibly huge imo

MasterTim17, 2019-08-09:
This commit changed the value for changing the value. I reverted this commit and it seems to help [https://github.com/MarlinFirmware/Marlin/commit/2d98bb39bd4cf0820fd3a0e6af33b1fdcae991c6](url)

closed 2019-08-10 by thinkyhead