MarlinFirmware/Marlin #14887: [BUG] Can't compile in PlatformIO + VSCode

klcjr89, 2019-08-08:
I was able to compile the recent commits from last night this morning, but not when I try to do the same I get the following error:

```> Executing task: platformio run <

The terminal shell path "platformio" does not exist

Terminal will be reused by tasks, press any key to close it.```

kirilldobr, 2019-08-08:
Same thing

kirilldobr, 2019-08-08:
Looks like the issue is with the PlatformIO itself.

larryyyyy, 2019-08-08:
have the same problem just now, with no reason. I was doing arduino thing. it worked for a long time but suddenly comes with this error. 

after some google, this maybe the reason. 
http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system 5

reinstall platformIO ide in vscode is not working for me. what i do is to install platformIO using pip, and it works as normal now. hope this help.

kirilldobr, 2019-08-08:
Yep, it helped. Pretty weird though.

larryyyyy, 2019-08-08:
> Yep, it helped. Pretty weird though.

I found a problem with this solution, I cannot use esp OTA to upload code now and it is so much slower in the OTA process...

RudyBenner, 2019-08-08:
delete
.pioenvs
.piolib
.piolibdeps
directors, allow it to rebuild. could take a while.
works here. New directory structure will look like this

[image]

larryyyyy, 2019-08-08:
> delete
> .pioenvs
> .piolib
> .piolibdeps
> directors, allow it to rebuild. could take a while.
> works here. New directory structure will look like this
> 
> [image]

this doesnt work in my situation. the platformio terminal has pio and platformio command. but the build and upload button (cmd + alt + b/u) are not working..

klcjr89, 2019-08-09:
Installing via pip also worked for me but not sure why that was the solution.

larryyyyy, 2019-08-09:
> this doesnt work in my situation. the platformio terminal has pio and platformio command. but the build and upload button (cmd + alt + b/u) are not working..

I finally solved the problem by removing "~/.platformio", vs code will build a new symlink afterward, and it works. No idea what is the problem. 

and about the esp ota. I updated the platformio esp32 lib to 1.9 which need more detail infomation for the connection, for example like port for ota and so on, which is different with the esp32 manufactor so it was not working if not correctly stated. it is now fixed.

closed 2019-08-09 by klcjr89

codincafe, 2019-08-09:
> have the same problem just now, with no reason. I was doing arduino thing. it worked for a long time but suddenly comes with this error.
>
> after some google, this maybe the reason. [...]

This worked for me. I guess VSCode has been updated which is messing up the workspace

tuxBurner, 2019-08-09:
I tried deleting ~/.platformio and ~/.vscode and the . directories in my project but still getting:

The terminal shell path "platformio" does not exist

hpsaturn, 2019-08-09:
I have the same issue after update my last version of vscode, but I tried too delete these directories but the issue persist, from console all stuff works fine

RudyBenner, 2019-08-09:
Should look like this. 

[image]

RudyBenner, 2019-08-09:
VSCODE 1.3.7 works fine here.

RudyBenner, 2019-08-09:
[image]

and 

[image]

tuxBurner, 2019-08-09:
Think it is related to: https://github.com/platformio/platformio-vscode-ide/issues/957

ph1nt, 2019-08-09:
Problem is in VSCode 1.37.0 on MacOS.
Solution for me:
1. delete '~/.platformio'
2. reinstall 'pip install platformio'
3. open VSCode and wait to rebuild configuration

Works too, but is uncomfortable:
1. open terminal inside VSCode
2. type platformio commands

jsiqueir, 2019-08-10:
pip install platformio on Mac 10.13.6 with VSCode 1.37.0 worked fine.
Thank you!

pliski, 2019-08-10:
> Problem is in VSCode 1.37.0 on MacOS.
This solved the problem also on Ubuntu 19.04.

> Problem is in VSCode 1.37.0 on MacOS.
> Solution for me:
> 
> 1. delete '~/.platformio'
> 2. reinstall 'pip install platformio'
> 3. open VSCode and wait to rebuild configuration
> 
> Works too, but is uncomfortable:
> 
> 1. open terminal inside VSCode
> 2. type platformio commands

This solved the problem also on Ubuntu 19.04.

joaodematejr, 2019-08-11:
@jsiqueir @pliski  
it worked, thank you very much

syproduction, 2019-08-11:
> Problem is in VSCode 1.37.0 on MacOS.
> Solution for me:
> 
> 1. delete '~/.platformio'
> 2. reinstall 'pip install platformio'
> 3. open VSCode and wait to rebuild configuration
> 
> Works too, but is uncomfortable:
> 
> 1. open terminal inside VSCode
> 2. type platformio commands

Worked on Mac OS 10.14.5

p3p, 2019-08-11:
Resolved with VSCode PlatformIO IDE extension 1.8.2

sl1pkn07, 2019-08-11:
if help, platformio vscode IDE have a option for use system instead of builtin piocore (downloaded throught platformio IDE)

in my macos, platformio is installed by homebrew, in my linux, with the package manager (archlinux/AUR), and in my windows, with PIP