MarlinFirmware/Marlin #14889: [BUG] Custom bootscreens not shown with MKS_MINI_12864 LCD

thisiskeithb, 2019-08-09:
### Description

Custom bootscreens are not shown when using the MKS Mini 12864 LCD (`MKS_MINI_12864`). This is similar to https://github.com/MarlinFirmware/Marlin/issues/13977, but is not specific to the SKR 1.3 (issue occurs on any hardware as long as the MKS Mini 12864 LCD is enabled.

### Steps to Reproduce

I'm testing with the [Tevo Michelangelo](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/config/examples/Tevo/Michelangelo) configs/MKS Gen-L specifically, but this occurs on any board with the `MKS_MINI_12864` controller enabled.

**Expected behavior:** Custom bootscreen to be displayed.

**Actual behavior:** Blank screen before Marlin bootscreen/info is displayed.

#### Additional Information

@DrDitto had proposed changes in PR https://github.com/MarlinFirmware/Marlin/pull/14216, but there were several comments related to their implementation of the fix & the PR was closed.

thisiskeithb, 2019-08-12:
This appears to also be related to https://github.com/MarlinFirmware/Marlin/issues/14174 and https://github.com/MarlinFirmware/Marlin/issues/13732 since defining the lcd as `MINIPANEL` has the same behavior.

thinkyhead, 2019-08-14:
If this change is reverted (plus any needed additions) is it better…?
```diff
- #define HAS_LCD_CONTRAST (HAS_GRAPHICAL_LCD && defined(DEFAULT_LCD_CONTRAST))
+ #define HAS_LCD_CONTRAST (                \
+      ENABLED(MAKRPANEL)                   \
+   || ENABLED(CARTESIO_UI)                 \
+   || ENABLED(VIKI2)                       \
+   || ENABLED(AZSMZ_12864)                 \
+   || ENABLED(miniVIKI)                    \
+   || ENABLED(ELB_FULL_GRAPHIC_CONTROLLER) \
+ )
```

thinkyhead, 2019-08-14:
Consolidating discussion over at #14174…

closed 2019-08-14 by thinkyhead