MarlinFirmware/Marlin #14890: [BUG] Pin number oversight for STM32F4x?

pspslimboy, 2019-08-09:
On my SKR Pro v1.1, trying to assign the PB5 pin results in the use of the PB4 pin instead. This is visible when using M43.

In the following file:
https://github.com/MarlinFirmware/Marlin/blob/2971b48a128662ab2f50a0a238aedfb64ec785d9/buildroot/share/PlatformIO/variants/BIGTREE_GENERIC_STM32F407_5X/variant.h#L111-L114
 
The PB4 and PB5 pins are set to the same number: 45. I am assuming this is an oversight because it causes issues with trying to use the PB5 pin, and the rest of the file is sequential. Setting the value of PB5 to 46 and viewing the pins again with M43 shows that it fixes this issue.

tpruvot, 2019-08-09:
yep, seems a typo :) someone forgot to put glasses ?

thisiskeithb, 2019-08-09:
See https://github.com/MarlinFirmware/Marlin/pull/14891

thisiskeithb, 2019-08-10:
The fix has been merged.

MangaValk, 2019-08-10:
My second extruder stopped working. Noticed in m122 it wouldn't activate the second extruder anymore. After this fix the second extruder does activate but is still not stepping. Guess there are more typos somewhere since it did work before

boelle, 2019-09-24:
will close as this has been fixed with #14891, we can always reopen if the problem is still there

PR #14891 (2019-08-10): Fix BigTreeTech STM32F407_5X / PB5 pin
### Description

The PB4 and PB5 pins are set to the same number: 45. This sets PB5 to 46.

### Related Issues

Fix #14890