MarlinFirmware/Marlin #14892: [BUG] not compiling skr e3 dip TMC2208

Bansh1990, 2019-08-09:
### compile errors

> Linking .pio\build\BIGTREE_SKR_MINI\firmware.elf
> .pio\build\BIGTREE_SKR_MINI\liba72\libTMCStepper.a(SW_UART.cpp.o): In function `SoftwareSerial::read()':
> c:\Marlin/.pio\libdeps\BIGTREE_SKR_MINI\TMCStepper\src\source/SW_UART.cpp:331: multiple definition of `SoftwareSerial::read()'
> .pio/build/BIGTREE_SKR_MINI/src/src/HAL/HAL_STM32F1/SoftwareSerial.cpp.o:c:\Marlin/Marlin\src\HAL\HAL_STM32F1/SoftwareSerial.cpp:44: first defined here
> .pio\build\BIGTREE_SKR_MINI\liba72\libTMCStepper.a(SW_UART.cpp.o): In function `SoftwareSerial::available()':
> c:\Marlin/.pio\libdeps\BIGTREE_SKR_MINI\TMCStepper\src\source/SW_UART.cpp:340: multiple definition of `SoftwareSerial::available()'
> .pio/build/BIGTREE_SKR_MINI/src/src/HAL/HAL_STM32F1/SoftwareSerial.cpp.o:c:\Marlin/Marlin\src\HAL\HAL_STM32F1/SoftwareSerial.cpp:40: first defined here
> .pio\build\BIGTREE_SKR_MINI\liba72\libTMCStepper.a(SW_UART.cpp.o): In function `SoftwareSerial::flush()':
> c:\Marlin/.pio\libdeps\BIGTREE_SKR_MINI\TMCStepper\src\source/SW_UART.cpp:362: multiple definition of `SoftwareSerial::flush()'
> .pio/build/BIGTREE_SKR_MINI/src/src/HAL/HAL_STM32F1/SoftwareSerial.cpp.o:c:\Marlin/Marlin\src\HAL\HAL_STM32F1/SoftwareSerial.cpp:50: first defined here
> .pio\build\BIGTREE_SKR_MINI\liba72\libTMCStepper.a(SW_UART.cpp.o): In function `SoftwareSerial::write(unsigned char)':
> c:\Marlin/.pio\libdeps\BIGTREE_SKR_MINI\TMCStepper\src\source/SW_UART.cpp:343: multiple definition of `SoftwareSerial::write(unsigned char)'
> .pio/build/BIGTREE_SKR_MINI/src/src/HAL/HAL_STM32F1/SoftwareSerial.cpp.o:c:\Marlin/Marlin\src\HAL\HAL_STM32F1/SoftwareSerial.cpp:48: first defined here
> .pio\build\BIGTREE_SKR_MINI\liba72\libTMCStepper.a(SW_UART.cpp.o): In function `SoftwareSerial::stopListening()':
> c:\Marlin/.pio\libdeps\BIGTREE_SKR_MINI\TMCStepper\src\source/SW_UART.cpp:133: multiple definition of `SoftwareSerial::stopListening()'
> .pio/build/BIGTREE_SKR_MINI/src/src/HAL/HAL_STM32F1/SoftwareSerial.cpp.o:c:\Marlin/Marlin\src\HAL\HAL_STM32F1/SoftwareSerial.cpp:57: first defined here
> .pio\build\BIGTREE_SKR_MINI\liba72\libTMCStepper.a(SW_UART.cpp.o): In function `SoftwareSerial::listen()':
> c:\Marlin/.pio\libdeps\BIGTREE_SKR_MINI\TMCStepper\src\source/SW_UART.cpp:114: multiple definition of `SoftwareSerial::listen()'
> .pio/build/BIGTREE_SKR_MINI/src/src/HAL/HAL_STM32F1/SoftwareSerial.cpp.o:c:\Marlin/Marlin\src\HAL\HAL_STM32F1/SoftwareSerial.cpp:53: first defined here
> collect2.exe: error: ld returned 1 exit status
> *** [.pio\build\BIGTREE_SKR_MINI\firmware.elf] Error 1

tpruvot, 2019-08-09:
i dont see any SW_UART.cpp in https://github.com/teemuatlut/TMCStepper/tree/master/src/source

Bansh1990, 2019-08-09:
I use this lib
https://github.com/bigtreetech/TMCStepper

> i dont see any SW_UART.cpp in https://github.com/teemuatlut/TMCStepper/tree/master/src/source

tpruvot, 2019-08-09:
ok, so.. try this for now...
https://github.com/MarlinFirmware/Marlin/pull/14893/files

Bansh1990, 2019-08-09:
change
platformio.ini 
[env:BIGTREE_SKR_MINI]
`-DDEBUG_LEVEL=0` 
to
`-DDEBUG_LEVEL=0 -DHAVE_SW_SERIAL`

and
Marlin\src\HAL\HAL_STM32F1\SoftwareSerial.cpp
19 string to
`#if defined(__STM32F1__) && !defined(HAVE_SW_SERIAL)`

compile with this fine. but cant check work on the board for now

tpruvot, 2019-08-09:
its a good start..

Bansh1990, 2019-08-10:
one more compile bug(

> Marlin\src\lcd\ultralcd.cpp: In static member function 'static void MarlinUI::update()':
> Marlin\src\lcd\ultralcd.cpp:767:9: error: 'touch_buttons' was not declared in this scope
>      if (touch_buttons) {
>          ^~~~~~~~~~~~~
> Marlin\src\lcd\ultralcd.cpp:767:9: note: suggested alternative: 'update_buttons'
>      if (touch_buttons) {
>          ^~~~~~~~~~~~~
>          update_buttons
> Marlin\src\lcd\ultralcd.cpp:805:5: error: 'else' without a previous 'if'
>      else wait_for_unclick = false;
>      ^~~~
> *** [.pio\build\BIGTREE_SKR_MINI\src\src\lcd\ultralcd.cpp.o] Error 1

sl1pkn07, 2019-08-10:
thats is other error unrelated to this bug

Bansh1990, 2019-08-11:
> change
> platformio.ini
> [env:BIGTREE_SKR_MINI]
> `-DDEBUG_LEVEL=0`
> to
> `-DDEBUG_LEVEL=0 -DHAVE_SW_SERIAL`

somebody can add this in pull request? this looks like a fix of compiling for skr mini and skr e3.
then it can be closed. thanx.

tpruvot, 2019-08-12:
the -DHAVE_SW_SERIAL is only when you add the lib, the custom TMCStepper lib from BIQ is not used in current platformio.ini

Bansh1990, 2019-08-12:
> the -DHAVE_SW_SERIAL is only when you add the lib, the custom TMCStepper lib from BIQ is not used in current platformio.ini

Okey. can be closed. thank you alot

closed 2019-08-12 by Bansh1990