MarlinFirmware/Marlin #14894: [BUG] Compiling M43 on invalid conversion

JohnnyTheOne, 2019-08-09:
Issue Compiling M43 on SKR PRO
get this error message
`<command-line>:0:0: note: this is the location of the previous definition
In file included from Marlin\src\gcode\config\../../pins/../HAL/HAL_STM32/pinsDebug.h:28:0,
                 from Marlin\src\gcode\config\../../pins/pinsDebug.h:101,
                 from Marlin\src\gcode\config\M43.cpp:29:
Marlin\src\gcode\config\../../pins/../HAL/HAL_STM32/pinsDebug_STM32duino.h: In function 'void port_print(pin_t)':
Marlin\src\gcode\config\../../pins/../HAL/HAL_STM32/pinsDebug_STM32duino.h:192:38: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
   char * const ppa = pin_xref[Index].Port_pin_alpha;
                                      ^~~~~~~~~~~~~~
*** [.pio\build\BIGTREE_SKR_PRO\src\src\gcode\config\M43.cpp.o] Error 1`

tpruvot, 2019-08-09:
@Bob-the-Kuhn @GMagician ....

GMagician, 2019-08-09:
`const char *` is different from `char const *` one is pointer const and other is data const (not sure if correct order) but I dont' know how `pin_xref[Index].Port_pin_alpha` is defined.
...is this the doubt?

Patag, 2019-08-09:
Fixed here by replacing
`char * const ppa = pin_xref[Index].Port_pin_alpha;`
by
  `const char *  ppa = pin_xref[Index].Port_pin_alpha;`

in Marlin-bugfix-2.0.x\Marlin\src\HAL\HAL_STM32\pinsDebug_STM32duino.h

JohnnyTheOne, 2019-08-09:
Patag, it works thanks

GMagician, 2019-08-09:
Please close

closed 2019-08-09 by JohnnyTheOne