MarlinFirmware/Marlin #14896: [BUG] Compile issue Marlin 2.0 for BIGTREETECH SKR Mini E3

Xenon-2510, 2019-08-09:
Description
Get  a compilation error when trying to compile for BIGTREETECH SKR MINI E3

Steps to Reproduce
Clone latest Marlin Github 2.0.bugfix
Replace configs with BIGTREETECH->SKR MINI E3->My Configuration files (attached)
Compile using enc:BIGTREE_SKR_MINI
Expected behavior:

Compiles correctly

Actual behavior:

In file included from c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\memory:62:0,
                 from C:\Users\arnoldc.BENTLEY\.platformio\lib\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:27,
                 from C:\Users\arnoldc.BENTLEY\.platformio\lib\EspSoftwareSerial_ID168\src/SoftwareSerial.h:27,
                 from .pio\libdeps\BIGTREE_SKR_MINI\TMCStepper_ID5513\src/TMCStepper.h:18,
                 from Marlin\src\module/stepper_indirection.h:47,
                 from Marlin\src\module/stepper.h:221,
                 from Marlin\src\Marlin.cpp:37:
c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:243:56: error: macro "min" passed 3 arguments, but takes just 2
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
                                                        ^
c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:265:56: error: macro "max" passed 3 arguments, but takes just 2
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
                                                        ^
Compiling .pio\build\BIGTREE_SKR_MINI\src\src\feature\Max7219_Debug_LEDs.cpp.o
In file included from c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\memory:62:0,
                 from C:\Users\arnoldc.BENTLEY\.platformio\lib\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:27,
                 from C:\Users\arnoldc.BENTLEY\.platformio\lib\EspSoftwareSerial_ID168\src/SoftwareSerial.h:27,
                 from .pio\libdeps\BIGTREE_SKR_MINI\TMCStepper_ID5513\src/TMCStepper.h:18,
                 from Marlin\src\core\../module/stepper_indirection.h:47,
                 from Marlin\src\core\../module/stepper.h:221,
                 from Marlin\src\core\utility.cpp:42:
c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:243:56: error: macro "min" passed 3 arguments, but takes just 2
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
                                                        ^
c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:265:56: error: macro "max" passed 3 arguments, but takes just 2
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
                                                        ^
In file included from c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algo.h:60:0,
                 from c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\algorithm:62,
                 from C:\Users\arnoldc.BENTLEY\.platformio\lib\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:28,
                 from C:\Users\arnoldc.BENTLEY\.platformio\lib\EspSoftwareSerial_ID168\src/SoftwareSerial.h:27,
                 from .pio\libdeps\BIGTREE_SKR_MINI\TMCStepper_ID5513\src/TMCStepper.h:18,
                 from Marlin\src\module/stepper_indirection.h:47,
                 from Marlin\src\module/stepper.h:221,
                 from Marlin\src\Marlin.cpp:37:
c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\algorithmfwd.h:375:41: error: macro "max" passed 3 arguments, but takes just 2
     max(const _Tp&, const _Tp&, _Compare);
                                         ^
c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\algorithmfwd.h:388:41: error: macro "min" passed 3 arguments, but takes just 2
     min(const _Tp&, const _Tp&, _Compare);
                                         ^
c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\algorithmfwd.h:416:30: error: macro "min" requires 2 arguments, but only 1 given
     min(initializer_list<_Tp>);
                              ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_SKR_MINI\src\src\Marlin.cpp.o] Error 1
In file included from c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\stl_algo.h:60:0,
                 from c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\algorithm:62,
                 from C:\Users\arnoldc.BENTLEY\.platformio\lib\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:28,
                 from C:\Users\arnoldc.BENTLEY\.platformio\lib\EspSoftwareSerial_ID168\src/SoftwareSerial.h:27,
                 from .pio\libdeps\BIGTREE_SKR_MINI\TMCStepper_ID5513\src/TMCStepper.h:18,
                 from Marlin\src\core\../module/stepper_indirection.h:47,
                 from Marlin\src\core\../module/stepper.h:221,
                 from Marlin\src\core\utility.cpp:42:
c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\algorithmfwd.h:375:41: error: macro "max" passed 3 arguments, but takes just 2
     max(const _Tp&, const _Tp&, _Compare);
                                         ^
c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\algorithmfwd.h:388:41: error: macro "min" passed 3 arguments, but takes just 2
     min(const _Tp&, const _Tp&, _Compare);
                                         ^
c:\users\arnoldc.bentley\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1\bits\algorithmfwd.h:416:30: error: macro "min" requires 2 arguments, but only 1 given
     min(initializer_list<_Tp>);
                              ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_SKR_MINI\src\src\core\utility.cpp.o] Error 1

I have searched for any information but cannot find anything conclusive, I did find a similar issue here: https://github.com/MarlinFirmware/Marlin/issues/11968

Not sure if this helps.

[Default Config.zip](https://github.com/MarlinFirmware/Marlin/files/3485845/Default.Config.zip)

tpruvot, 2019-08-09:
see #14892

Xenon-2510, 2019-08-09:
I have made the changes referenced in #14892 and still have the same error when compiling

tpruvot, 2019-08-09:
you installed EspSoftwareSerial which is not arduino compaible (mbed only)

tpruvot, 2019-08-09:
There is none compaible listed in pio database yet... Need the fysetc SoftwareSerialM or the bigtree TMCStepper lib which include one... on the STM32F1

Bansh1990, 2019-08-09:
> I have made the changes referenced in #14892 and still have the same error when compiling

try this

>

>  [common]
> default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
> build_flags = -fmax-errors=5
>   -g
>   -ggdb
> lib_deps =
>   U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
>   LiquidCrystal@1.3.4
>   https://github.com/bigtreetech/TMCStepper
>   Adafruit NeoPixel@1.1.3
>   LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
>   Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/dev.zip
>   TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
>   SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
>   SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip
>   SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip

tpruvot, 2019-08-09:
no no... i see no link to https://github.com/bigtreetech/TMCStepper there

Bansh1990, 2019-08-09:
> no no... i see no link to https://github.com/bigtreetech/TMCStepper there

look carefully in [common] ) 8 string)

tpruvot, 2019-08-09:
i overheat, sorry ^^

reloxx13, 2019-08-09:
i use a mini e3 myself with latest marlin and it compiles fine.

clean ur lib folders which are in ur marlin folder and not in git.

reloxx13, 2019-08-09:
and u cannot copy the btt configs cause they almost always out of day, if u want take my config files as a start.

 u can find my config here: https://github.com/reloxx13/Marlin/tree/my-skr-mini-e3/Marlin

i merged them manually with git diffs

Bansh1990, 2019-08-09:
confirm compile without `-DHAVE_SW_SERIAL`  on skr e3 dip

Xenon-2510, 2019-08-09:
Thank you so much, I have uninstalled the ESPSoftwareSerial and it has compiled successfully. 

@reloxx13 - Thanks, I will try your configs anyway as I see that you are using a BLTouch and I have been having intermittent issues with the probe triggering.

closed 2019-08-09 by Xenon-2510