MarlinFirmware/Marlin #14898: [BUG] (2.0) does MINIMUM_STEPPER_DIR_DELAY work ?

maukcc, 2019-08-09:
We are having problems with getting babystepping to work both ways. It will only go in the direction that the previous move was in.
So we were thinking that could be an issue with timing.
Then we set the MINIMUM_STEPPER_DIR_DELAY to a ridiculous amount of 2000000000 (2sec)
Then we expected a delay of 2 sec. between a direction change.
That did not happen.
We have no driver types selected
Do we need to do anything else (except un-commenting the line)

maukcc, 2019-08-14:
We also tried with #define X_DRIVER_TYPE  TB6600 and MINIMUM_STEPPER_DIR_DELAY set to 10000000000 (10sec).
But instant directional changes, no 10 sec wait

gloomyandy, 2019-08-14:
I'm not sure that your way of testing is going to work. I'm pretty sure that the delay period will be being stored as a 32bit value and 10,000,000,000 (10 sec)is larger than  the max value that can be stored in a 32 bit unsigned int (4,294,967,295) or a 32 bit signed int (2,147,483,647).

rusnob, 2019-08-19:
There is a problem in marlin

my comments on this

https://github.com/MarlinFirmware/Marlin/issues/13559

https://github.com/MarlinFirmware/Marlin/issues/13485

closed 2019-09-23 by maukcc