MarlinFirmware/Marlin #14899: [FR] M108 breaks out of an M109 or M190 wait-for-temperature loop

maukcc, 2019-08-09:
The M108 breaks out of an M109 or M190 wait-for-temperature loop, but it still keeps the temp in the loop.
So if you set M190 S90, then realize it should be 50, you will have to wait until it gets to 90.
Can this be changed?

reloxx13, 2019-08-09:
this?
```
M190 S90
M108
M190 S50
```

maukcc, 2019-08-09:
That will still wait for the temp to reach 90, then sets it to 50.

shitcreek, 2019-08-09:
Do you have EMERGENCY_PARSER enabled?

maukcc, 2019-08-09:
Yes, the M108 works. So The M190 S90 loop gets ended and it continieus with printing. We just can not change the bed temp untill it reaches 90.

GMagician, 2019-08-09:
Are you using last 2.0 bugfix?

maukcc, 2019-08-09:
No not latest, couple of months old 2.0

GMagician, 2019-08-09:
Not sure it will change something but maybe try the newest...I checked code and it seems that M108 should break m190 loop

maukcc, 2019-08-09:
Just to be clear:
yes, it breaks the loop and continues printing. 
No, temps can not be changed until it reaches target(while it has gone on with printing)

GMagician, 2019-08-09:
not clear...are you printing? so when M108 and next M190 is sent....are they part of gcode?
M190 seems to set new target everytime it's executed with s parameter

maukcc, 2019-08-12:
We can not reproduce the error any more.
Guess it got stuck in the M109 loop that came right after the M190 code

closed 2019-08-12 by maukcc