MarlinFirmware/Marlin #14901: [BUG] compiling error service interval with watchdogs

Ludy87, 2019-08-09:
### Description

Wont to compile by enbale `PRINTCOUNTER` and defined `SERVICE_NAME_1` & `SERVICE_INTERVAL_1` -> compiling error

```c++
Marlin\src\lcd\menu\menu_service.cpp: In lambda function:
Marlin\src\lcd\menu\menu_service.cpp:40:44: error: 'index' is not captured
       print_job_timer.resetServiceInterval(index);
                                            ^
Marlin\src\lcd\menu\menu_service.cpp:39:6: note: the lambda has no capture-default
     []{
      ^
Marlin\src\lcd\menu\menu_service.cpp:34:37: note: 'const int index' declared here
 inline void _menu_service(const int index, PGM_P const name) {
```

### Steps to Reproduce

1. `#define PRINTCOUNTER`
2. `#define SERVICE_NAME_1      "Service S"`
3. `#define SERVICE_INTERVAL_1  100`

**Expected behavior:** compiled

**Actual behavior:** do not compiled

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3487184/Marlin.zip)

Patag, 2019-08-09:
Duplicate of #13816

Ludy87, 2019-08-09:
Thanks @Patag 

*25 Apr*

closed 2019-08-09 by Ludy87