MarlinFirmware/Marlin #14902: [Question] LiquidCrystal

GMagician, 2019-08-09:
Why in platformio there is a bond with version? I checked latest downloaded by pio and the only remarkable difference is a "analogwrite" funciton. Removed forced version it compiles fine...

Same for adafruitneopixel (but I have to check differences here)

GMagician, 2019-08-09:
neopixel new lib seems to have a different approch for samd51...but memory footprint seems bigger

Ludy87, 2019-08-09:
why keep up-to-date if there is no error? Or are there new features?

Never change a running system! :-)

GMagician, 2019-08-09:
if you say pio update project library you will get a warning then, about duplicated libraries

neopixel for sure has something different somewhere... a [++index] vs [index++] may do differences (don't checked yet starting index value) and for sure it support samd51 (even if with some hack on SysTick)

GMagician, 2019-08-09:
> why keep up-to-date if there is no error

for the same reason you update marlin to new version :-D

Ludy87, 2019-08-09:
I do not update Marlin :-P

That is of course one argument

GMagician, 2019-08-09:
Joking aside, this new libs doesn't fix anything, it adds only support to esp32

Ludy87, 2019-08-09:
New function is a argument 😊

GMagician, 2019-08-10:
New newpixel library is required by samd51 to support 400KHz...current library doesn't support it. Also
current version uses "nop" to realize duty, new "disable" interrupts and uses Systick...with some hacks that need to be considered (millis will lose some amount of time)...what I can do?

GMagician, 2019-08-27:
I don't use neopixel so I'll not care about this "issue" until someone will point it out.

closed 2019-08-27 by GMagician