MarlinFirmware/Marlin #14904: [Question] duplicate macro LCD vs SAMD51

GMagician, 2019-08-09:
I'm going to see what is needed to add lcd to samd51 and first thing I see is a macro redefinition error...
`#define WIDTH 128` in lcd collide with a struct field

```c++
struct {
    uint32_t WIDTH:3;    
```

I don't understand why this occurs (2nd time a I face a issue with macro and struct fields, it make non sens to me that a field is replaced by a macro but this seems to be)

what can be good solution to solve this since all of them are outside marlin code?

GMagician, 2019-08-27:
@thinkyhead some decision has to be taken to solve this if lcd on samd51 is required on main stream.
I also wrote to ug8 repository but no answer even there...
Since lcd lib is custom to Marlin I think its better to change it, not generic samd51 lib, but somewhere a change has to be done...last option is to have my personal LCD lib and leave main stream without lcd (on samd)

GMagician, 2019-08-29:
Work in progress...

closed 2019-08-29 by GMagician

thinkyhead, 2019-08-29:
The error was caused by one of the u8g LCD implementations defining `WIDTH` before including `Arduino.h` (or `WProgram.h`). I have patched `clib/u8g_dev_ht1632.c` so it now has the includes before the define, so only the C file sees it.