MarlinFirmware/Marlin #14905: [BUG] When #define SPI_ENDSTOPS compile fails

McVillano, 2019-08-09:
### Description

When trying to compile marlin 2.0.x bugfix for the SKR 1.3 with LPC1768 platform and platform.io fails if i set  #define SPI_ENDSTOPS for not wiring diag to endstop. It prints 

### Steps to Reproduce

1. use platformio.ini attached 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3487661/Marlin.zip)

2. Use configuration.h and configuration_adv.h attached
3. Compile

**Actual behavior:**

```
In file included from Marlin\src\inc/MarlinConfigPre.h:34,
                 from Marlin\src\inc/MarlinConfig.h:28,
                 from Marlin\src\Marlin.h:24,
                 from Marlin\src\Marlin.cpp:31:
Marlin\src\Marlin.cpp: In function 'void idle(bool)':
Marlin\src\Marlin.cpp:664:58: error: 'sg_guard_period' was not declared in this scope
     if (endstops.tmc_spi_homing.any && ELAPSED(millis(), sg_guard_period))
                                                          ^~~~~~~~~~~~~~~
Marlin\src\inc/../core/millis_t.h:28:43: note: in definition of macro 'PENDING'
 #define PENDING(NOW,SOON) ((int32_t)(NOW-(SOON))<0)
                                           ^~~~
Marlin\src\Marlin.cpp:664:40: note: in expansion of macro 'ELAPSED'
     if (endstops.tmc_spi_homing.any && ELAPSED(millis(), sg_guard_period))
                                        ^~~~~~~
*** [.pio\build\LPC1768\src\src\Marlin.cpp.o] Error 1
```

GMagician, 2019-08-09:
Not sure because I don't know how logic works but `sg_guard_period` is valid only when `IMPROVE_HOMING_RELIABILITY` is enabled, maybe just write
`if (endstops.tmc_spi_homing.any)`
but I think @teemuatlut may say more than me

sl1pkn07, 2019-08-09:
Hi

i can confirm disable IMPROVE_HOMING_RELIABILITY makes build fail (ReARM)

greetings

GMagician, 2019-08-10:
Fixed and merged..please close

McVillano, 2019-08-10:
I've re-donwloaded the source code from the webpage but still the same error:
```
Compiling .pio\build\LPC1768\src\src\feature\bedlevel\bedlevel.cpp.o
In file included from Marlin\src\inc/MarlinConfigPre.h:34,
                 from Marlin\src\inc/MarlinConfig.h:28,
                 from Marlin\src\Marlin.h:24,
                 from Marlin\src\Marlin.cpp:31:
Marlin\src\Marlin.cpp: In function 'void idle(bool)':
Marlin\src\Marlin.cpp:664:58: error: 'sg_guard_period' was not declared in this scope
     if (endstops.tmc_spi_homing.any && ELAPSED(millis(), sg_guard_period))
                                                          ^~~~~~~~~~~~~~~
Marlin\src\inc/../core/millis_t.h:28:43: note: in definition of macro 'PENDING'
 #define PENDING(NOW,SOON) ((int32_t)(NOW-(SOON))<0)
                                           ^~~~
Marlin\src\Marlin.cpp:664:40: note: in expansion of macro 'ELAPSED'
     if (endstops.tmc_spi_homing.any && ELAPSED(millis(), sg_guard_period))
                                        ^~~~~~~
Compiling .pio\build\LPC1768\src\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp.o
*** [.pio\build\LPC1768\src\src\Marlin.cpp.o] Error 1
```

sl1pkn07, 2019-08-10:
for me is fixed with code from git

GMagician, 2019-08-10:
` if (endstops.tmc_spi_homing.any && ELAPSED(millis(), sg_guard_period))`

this code doesn't come from merge, this the same old code

McVillano, 2019-08-10:
You are right, i was downloading old version. 

I have another error now but it's an UltraLCD thing THANKS!

closed 2019-08-10 by McVillano

GMagician, 2019-08-10:
> I have another error now but it's an UltraLCD

maybe related to #14912?

McVillano, 2019-08-10:
> > I have another error now but it's an UltraLCD
> 
> maybe related to #14912?

Sure, the same, jejejeje you are a magician, but it's extrange to me because i don´t use ultraLCD i have it disabled and i use a REPRAP_DISCOUNT_SMART_CONTROLLER.

tpruvot, 2019-08-10:
I think Scott overheated and made that to reduce the PRs amount over the week end :p

PR #14906 (2019-08-09): Fix SPI_ENDSTOPS compile error
Fix #14905

Before merge @teemuatlut should confirm this fix is correct