MarlinFirmware/Marlin #14910: [Question] How should DELTA_DIAGONAL_ROD_TRIM_TOWER be used for calibration?

jackw01, 2019-08-10:
I can't seem to find any documentation on what DELTA_DIAGONAL_ROD_TRIM_TOWER does, other than that it's a way to compensate for unequal diagonal rod lengths that can cause dimensional accuracy. I was having dimensional accuracy issues on my delta printer running Marlin 1.1.8 and tried calibrating using these values using methods documented [on Thingiverse](https://www.thingiverse.com/thing:745523), but it did not seem to affect my prints. I started using a trial-and-error approach to find the correct values instead and arrived at { 0.0, -4.798, 15.114 } for DELTA_DIAGONAL_ROD_TRIM_TOWER which gave correctly sized prints. This would imply that my longest rod is ~20mm longer than the shortest one, but I measured them the best I could and they are all between 180 and 181mm. Because of this, it's not clear to me what DELTA_DIAGONAL_ROD_TRIM_TOWER actually represents, how it is supposed to be calibrated, or if there is something else wrong with my printer.

mylife4aiurr, 2019-08-12:
delta user, subbing 4 answer

jackw01, 2019-08-13:
After a lot of experimentation, I ended up arriving at a DELTA_DIAGONAL_ROD_TRIM_TOWER value of { 6.348, -2.297, -3.480 } which seems to give dimensionally accurate prints for me. However, this suggests that my longest rod is over 9mm longer than the shortest one, but they are actually all between 180 and 181mm long. The endstop, tower angle, radius, and height values in M665 that I obtained through auto calibration all accurately reflect the dimensions of my printer.

boelle, 2019-09-24:
@jackw01 better ask in discord

-Marlin Discord server. Join link: https://discord.gg/n5NJ59y

boelle, 2019-10-24:
better ask here:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

this list is for bugs and pr's only

closed 2019-10-24 by boelle

lkvapil, 2020-02-11:
It's so weird, I'm calibrating my delta nearly for two months. I mean when I set DELTA_DIAGONAL_ROD_TRIM_TOWER and print dimension test on each tower A,B and the C, it's correct but when I print object to calibrate them all at time, it get's wrong dimensions.

Any thoughts ? 
Endstops and print radius are calibrated well..

Thank you

thinkyhead, 2020-02-11:
> Any thoughts ?

I've never had a lovable delta, so all I know is that [`G33` is used](https://marlinfw.org/docs/gcode/G033.html) to figure this stuff out.

lkvapil, 2020-02-11:
but G33 doesn't solver dimension accuracy or does ? 
my delta is well built, sturdy joints so I do not know why it print's inaccurately :/

SKR1.3, was using waterott's TMC2208, now LV8729 with 128 stepping

thinkyhead, 2020-02-11:
> but G33 doesn't solver dimension accuracy or does ?

🤷‍♂ As I said, that's "all I know."

lkvapil, 2020-02-11:
thank you for your effort, otherwise do you know someone who could help me ?

thinkyhead, 2020-02-11:
A few thousand people.

‪- http://marlinfw.org‬
‪- http://forums.reprap.org/list.php?415‬
‪- https://discuss.toms3d.org/‬
‪- https://www.facebook.com/groups/1049718498464482/‬
‪- https://www.facebook.com/groups/3Dtechtalk/‬
‪- https://www.youtube.com/results?search_query=marlin+configuration‬
‪- https://discord.gg/n5NJ59y‬