MarlinFirmware/Marlin #14916: [BUG] Can't compile for LPC1769

klcjr89, 2019-08-10:
The build is broken again and I cannot compile in VSCode / PlatformIO. I receive the following errors:

```
Marlin/src/lcd/ultralcd.cpp: In static member function 'static void MarlinUI::update()':
Marlin/src/lcd/ultralcd.cpp:767:9: error: 'touch_buttons' was not declared in this scope
     if (touch_buttons) {
         ^~~~~~~~~~~~~
Marlin/src/lcd/ultralcd.cpp:767:9: note: suggested alternative: 'buttons'
     if (touch_buttons) {
         ^~~~~~~~~~~~~
         buttons
Marlin/src/lcd/ultralcd.cpp:805:5: error: 'else' without a previous 'if'
     else wait_for_unclick = false;
     ^~~~
Compiling .pio/build/LPC1769/src/src/libs/nozzle.cpp.o
Compiling .pio/build/LPC1769/src/src/libs/numtostr.cpp.o
*** [.pio/build/LPC1769/src/src/lcd/ultralcd.cpp.o] Error 1
============== [ERROR] Took 117.38 seconds ==============
```

Thank you!

sl1pkn07, 2019-08-10:
https://github.com/MarlinFirmware/Marlin/pull/14912

asus-zenfone6, 2019-08-10:
I have the same issue. Had to revert back to previous versions of ultralcd.h and ultralcd.cpp files.

closed 2019-08-11 by klcjr89