MarlinFirmware/Marlin #14921: [BUG] Wrong description for the TMC2209 sensorless homing

holgin, 2019-08-10:
Hello,

In the Configuration_adv.h file, description explaining how to set `AXIS_STALL_SENSITIVITY`  (starting line 1853) is wrong for the TMC2209 drivers, because the **higher** the value, the **more** sensitive StallGuards is. It's a really small error but I've wasted some time figuring it out.

biskero, 2019-08-10:
I was confused about this too since some people are stating the opposite and [here](http://marlinfw.org/docs/gcode/M914.html) is defined for TMC 2130 as **Higher values = Lower sensitivity**.
So the documents should be updated as you suggested.

holgin, 2019-08-10:
I've played with values aroung 80...120, but now I'm lost. At this moment I am unable to trigger SG (high pulse on DIAG pin), no matter the sensitivity, load, speed, even power supply voltage...

Patag, 2019-08-10:
@holgin don't forget do initialize/save settings in Eeprom to be sure your new ones defined in configuration_adv.h are really taken in account.

holgin, 2019-08-10:
@Patag I monitor all settings with M122, the board is brand new and nothing is stored in EEPROM yet. But I got it right:
-With `Y_STALL_SENSITIVITY` set to 250, or any value over ~150 actually, the driver pulses HIGH on DIAG pin immediately after starting a move.
-With value between 80...120, the sensitivity is reasonable.
-With `Y_STALL_SENSITIVITY` set to 20, I can block the motor and despite it losing all the steps, SG is not triggered and the homing fails.
 
The SG works only when homing, I think, that's why I could not get it to trigger while doing tests with 'G1' moves. During my first tests SG _was_ triggering during normal moves, not sure why though...

But, to conclude - the description is wrong, I've confirmed it, nothing serious but worth correcting.

lokus77, 2019-08-23:
tmc2209 datasheet "SGTHRS" register info says: 
Figure 11.1 Function principle of StallGuard4
**SGTHRS**:  This value controls the StallGuard4 threshold level for stall detection. It compensates for
motor specific characteristics and controls sensitivity. 
A higher value gives a higher sensitivity. A higher value makes StallGuard4 more sensitive and requires less torque to indicate a stall.

teemuatlut in stallguard examples  (here: https://github.com/teemuatlut/TMCStepper/blob/master/examples/StallGuard_TMC2209/StallGuard_TMC2209.ino )  says:
*A higher STALL_VALUE will make the reading less sensitive and
*A lower STALL_VALUE will make it more sensitive.
but he uses" SGTHRS"  -->   driver.SGTHRS(STALL_VALUE); 100 by default in example.

thisiskeithb, 2019-08-29:
PR https://github.com/MarlinFirmware/Marlin/pull/15092 clarifies TMC2209 sensitivity settings.

biskero, 2019-08-29:
it would be good to add this info to the Marlin docs page:
http://marlinfw.org/docs/gcode/M914.html

thisiskeithb, 2019-08-29:
> it would be good to add this info to the Marlin docs page:
> http://marlinfw.org/docs/gcode/M914.html

@shitcreek: Mind updating that page for TMC2209s?

thisiskeithb, 2019-08-29:
These changes have been merged! The new description in `Configuration_adv.h` reads as follow:
```c++
   * X/Y/Z_STALL_SENSITIVITY is the default stall threshold.
   * Use M914 X Y Z to set the stall threshold at runtime:
   *
   *  Sensitivity   TMC2209   Others
   *    HIGHEST       255      -64    (Too sensitive => False positive)
   *    LOWEST         0        63    (Too insensitive => No trigger)
```

thisiskeithb, 2019-08-29:
@boelle: This issue can be closed since the description for 2209s has been updated.

boelle, 2019-08-30:
oki, will close
we can reopen if major protests rise :-D

closed 2019-08-30 by boelle