MarlinFirmware/Marlin #14927: PROBLEM WITH A COMPILATION MARLIN 2.0.X ON RE-ARM

Rubenazzzo, 2019-08-11:
I adquired a re-arm board and i have a issues with the compilation, (I use the ATOM software), y follow a tutorials.
I search on the web about of this problem and on this issues, i cant foun nothing.
Sorry for my english.
When i go to create the PIO Building, the program show me this erros, I search on the directory , i see all are normal,
This are the problems with the pins (is possible is clear the showme , but  i cant found):
 

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.0.2 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.80KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: framework-arduino-lpc176x 0.1.3, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino

LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.4.6
|-- <Adafruit NeoPixel> 1.2.4
|-- <SailfishLCD>
 
Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upl
oad.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfig.h:32:0,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp:26:
Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../pins/pins.h:537:4: error: #error "Unknown MOTHERBOARD value set in Configuration.h"
   #error "Unknown MOTHERBOARD value set in Configuration.h"
    ^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:32:0,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:537:4: error: #error "Unknown MOTHERBOARD value set in Configuration.h"
   #error "Unknown MOTHERBOARD value set in Configuration.h"
    ^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp:26:
Marlin\src\HAL\HAL_LPC1768\../../core/../inc/SanityCheck.h:1457:4: error: #error "HEATER_0_PIN not defined for this board."

   #error "HEATER_0_PIN not defined for this board."
    ^~~~~
Marlin\src\HAL\HAL_LPC1768\../../core/../inc/SanityCheck.h:1599:8: error: #error "E0_STEP_PIN, E0_DIR_PIN, or E0_ENABLE_PIN not def
ined for this board."
       #error "E0_STEP_PIN, E0_DIR_PIN, or E0_ENABLE_PIN not defined for this board."
        ^~~~~
Marlin\src\HAL\HAL_LPC1768\../../core/../inc/SanityCheck.h:1603:10: error: #error "E1_STEP_PIN, E1_DIR_PIN, or E1_ENABLE_PIN not de
fined for this board."
         #error "E1_STEP_PIN, E1_DIR_PIN, or E1_ENABLE_PIN not defined for this board."
          ^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1457:4: error: #error "HEATER_0_PIN not defined for this board."
   #error "HEATER_0_PIN not defined for this board."
    ^~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1599:8: error: #error "E0_STEP_PIN, E0_DIR_PIN, or E0_ENABLE_PIN not defined for
 this board."
       #error "E0_STEP_PIN, E0_DIR_PIN, or E0_ENABLE_PIN not defined for this board."
        ^~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1603:10: error: #error "E1_STEP_PIN, E1_DIR_PIN, or E1_ENABLE_PIN not defined fo
r this board."
         #error "E1_STEP_PIN, E1_DIR_PIN, or E1_ENABLE_PIN not defined for this board."
          ^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:32:0,
                 from Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp:51:

Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:537:4: error: #error "Unknown MOTHERBOARD value set in Configuration.h"
   #error "Unknown MOTHERBOARD value set in Configuration.h"
    ^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:32:0,
                 from Marlin\src\HAL\HAL_LPC1768\HAL_timers.cpp:31:
Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/pins.h:537:4: error: #error "Unknown MOTHERBOARD value set in Configuration.h"
   #error "Unknown MOTHERBOARD value set in Configuration.h"
    ^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp:51:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1457:4: error: #error "HEATER_0_PIN not defined for this board."
   #error "HEATER_0_PIN not defined for this board."
    ^~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1599:8: error: #error "E0_STEP_PIN, E0_DIR_PIN, or E0_ENABLE_PIN not defined for
 this board."
       #error "E0_STEP_PIN, E0_DIR_PIN, or E0_ENABLE_PIN not defined for this board."
        ^~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1603:10: error: #error "E1_STEP_PIN, E1_DIR_PIN, or E1_ENABLE_PIN not defined fo
r this board."
         #error "E1_STEP_PIN, E1_DIR_PIN, or E1_ENABLE_PIN not defined for this board."
          ^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_LPC1768\HAL_timers.cpp:31:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1457:4: error: #error "HEATER_0_PIN not defined for this board."
   #error "HEATER_0_PIN not defined for this board."

    ^~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1599:8: error: #error "E0_STEP_PIN, E0_DIR_PIN, or E0_ENABLE_PIN not defined for
 this board."
       #error "E0_STEP_PIN, E0_DIR_PIN, or E0_ENABLE_PIN not defined for this board."
        ^~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1603:10: error: #error "E1_STEP_PIN, E1_DIR_PIN, or E1_ENABLE_PIN not defined fo
r this board."
         #error "E1_STEP_PIN, E1_DIR_PIN, or E1_ENABLE_PIN not defined for this board."
          ^~~~~
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_spi.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.cpp.o] Error 1
========================== [ERROR] Took 5.20 seconds ==========================
 
================================== [SUMMARY] ==================================
Environment megaatmega2560              [IGNORED]
Environment megaatmega1280              [IGNORED]
Environment at90usb1286_cdc             [IGNORED]
Environment at90usb1286_dfu             [IGNORED]
Environment DUE                         [IGNORED]
Environment DUE_USB                     [IGNORED]
Environment DUE_debug                   [IGNORED]
Environment LPC1769                     [IGNORED]
Environment melzi                       [IGNORED]
Environment melzi_optiboot              [IGNORED]
Environment rambo                       [IGNORED]
Environment sanguino_atmega644p         [IGNORED]
Environment sanguino_atmega1284p        [IGNORED]

Environment STM32F1                     [IGNORED]
Environment fysetc_STM32F1              [IGNORED]
Environment BIGTREE_SKR_MINI            [IGNORED]
Environment STM32F4                     [IGNORED]
Environment STM32F7                     [IGNORED]
Environment ARMED                       [IGNORED]
Environment alfawise_U20                [IGNORED]
Environment mks_robin                   [IGNORED]
Environment mks_robin_lite              [IGNORED]
Environment mks_robin_mini              [IGNORED]
Environment mks_robin_nano              [IGNORED]
Environment jgaurora_a5s_a1             [IGNORED]
Environment black_stm32f407ve           [IGNORED]
Environment BIGTREE_SKR_PRO             [IGNORED]
Environment teensy31                    [IGNORED]
Environment teensy35                    [IGNORED]
Environment malyanm200                  [IGNORED]
Environment esp32                       [IGNORED]
Environment fysetc_f6_13                [IGNORED]
Environment linux_native                [IGNORED]
Environment adafruit_grandcentral_m4    [IGNORED]
Environment LPC1768                     [FAILED]
==================== 1 failed, 0 succeeded in 5.21 seconds ====================

thisiskeithb, 2019-08-11:
A few things:

1. Zip up your `Configuration.h` & `Configuration_adv.h` files and attach them as the [issue template](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/issue_template.md) requested. This can help us troubleshoot the `Unknown MOTHERBOARD value set in Configuration.h` error.
3. Make sure you're using the latest `bugfix-2.0.x` code.
2. PlatformIO is no longer supported in Atom, so use [VSCode](https://docs.platformio.org/en/latest/ide/vscode.html).

Rubenazzzo, 2019-08-11:
Thanks, for your quikly answer.
Now are out if te House but are easy, when y see the problem, download the latest versión, and configure 2 extrusor and a single nozzle,termisstor 1 in a bed and noozle,the others a 0,basic configuration for chek again.

About use the vscode, when arrive try with them,again.

Regards
Un saludo.

De: thisiskeithb
Enviado: domingo, 11 de agosto 11:03
Asunto: Re: [MarlinFirmware/Marlin] PROBLEM WITH A COMPILATION MARLIN 2.0.X ON RE-ARM (#14927)
Para: MarlinFirmware/Marlin
Cc: Rubenazzzo, Author

A few things:
Zip up your Configuration.h & Configuration_adv.h files and attach them as the issue template<https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/issue_template.md> requested. This can help us troubleshoot the Unknown MOTHERBOARD value set in Configuration.h error.Make sure you're using the latest bugfix-2.0.x code.PlatformIO is no longer supported in Atom, so use VSCode<https://docs.platformio.org/en/latest/ide/vscode.html>.

Rubenazzzo, 2019-08-11:
So sorry for the wait, I've been out all day.
I have done what you told me, downloaded the latest version of marlin, installed the Visual Studio Code, and reconfigured the configuration.h / configuration_adv.h files.
The result is the same, I can't compile.
I upload the configured marlin (it currently works with that configuration in an Atmega + ramps)
and a text file with the wrong compilation.
Thanks in advance for your help.

[Marlin-bugfix-2.0.x_11-08-19_mod.zip](https://github.com/MarlinFirmware/Marlin/files/3490369/Marlin-bugfix-2.0.x_11-08-19_mod.zip)
[compiled_failed_11-08-19_2140.txt](https://github.com/MarlinFirmware/Marlin/files/3490370/compiled_failed_11-08-19_2140.txt)

thisiskeithb, 2019-08-12:
Try defining your motherboard as `BOARD_RAMPS_14_RE_ARM_EEB` instead of `BOARD_RAMPS_RE_ARM_EEB`.

Rubenazzzo, 2019-08-12:
I can't believe it, it was that, I am sure that initially and in one of the tests, I literally copied it from the boards file.
I do not know how to thank you.
It is true that many times a simple thing does not let you see the world.
Thank you all.
;)

tpruvot, 2019-08-12:
remember to close the issue ;)

closed 2019-08-12 by Rubenazzzo