MarlinFirmware/Marlin #14928: [BUG] TMC2130 Only work sensorless homing in Y axis, X axis Failed (never stop)

sl1pkn07, 2019-08-11:
### Description

Enable sensorless homing (diag1 wired) in X and Y axis, only work the Y axis, the X axis never stop/detect endstop

things i try:

-Switch/Replace Diag1 Wires
-Switch X and Y driver (mount X driver in Y axis, mount Y driver in X axis)
-Replace X axis with brand new driver
-Using a mechanical endtop in X works, then is not problem of the endstop pins iself
-Change the X stepmotor with other and try to emulate the endstop with my hand. the motor loses steps but never stop until the printer halted
-Try all X axis range of sensitivity, from -64 yo 63

### Steps to Reproduce

1. Enable `SENSORLESS_HOMING` and set the sensitivity
2. Weld a pin in the Diag1 header in the TMC1230
3. Conect each (X and Y driver) Diag1 pin to endtop header with dupont wire
4. Try to Homing (from LCD or pronterface/esp3d), all axis, only X and only Y

**Expected behavior:** [What you expect to happen]

X and Y axis homing sucess

**Actual behavior:** [What actually happens]

Y axis sucess
X axis never stop when hit the limit. if not poweroff the machine (the sound is horribe), the machine halt in some seconds

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

https://github.com/sl1pkn07/Marlin/commit/eb84d632328138664a0317a6cedf697e7ecc7e93

* Provide pictures or links to videos that clearly demonstrate the issue.

when charge the phone

* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

GMagician, 2019-08-11:
don't know if it's the problem but in your configuration you have TMC2160

sl1pkn07, 2019-08-11:
ops!, you are right, my fault, but change to 2130 do nothing. same issue in X

sl1pkn07, 2019-08-11:
the vid

https://youtu.be/D1HMqXxp-nc

sorry for the koality

sl1pkn07, 2019-08-12:
more test:

-Remap (one by one) the endstop pins as Y_min pin for test Y axis with all endstop pins : ALL SUCCESS
-Remap (one by one) the endstop pins as X_min pin for test X axis with all enstop pins: ALL FAILED
-with DMM, meassure between the diag1 pins on X and Y, and GND (one by one) when do axis homing. both show 2,962v (more or less), when homing Y, the voltage down +/-50mV (+/-2,912v), but not X axis (untouch voltage)

also i've reweld all pines in the realm for if is a problem with a cold weld (dat fuc.... rohs compilant)

the problem persist

i have a taurino (mega 24v) for test again

greetings

sl1pkn07, 2019-08-13:
other test

remap all X stepper pins to physical E1 pins: same issue (with 3 different drivers)

Bob-the-Kuhn, 2019-08-13:
Another possibility - maybe there's no SPI communication with the X driver.  Try changing `#define X_MICROSTEPS ` in the TMC section and see if that works.

sl1pkn07, 2019-08-13:
FFFFFFFFFFFFFFFFFFFffffffffffffffffff

seems a SPI wired kit brougth (biqu) for don't make myself ones is the problem, change for other now all fixed

all wires give continuity by MMD, but one terminal seems get problems with SPI. i not sure why

sorry :/

greetings and close

closed 2019-08-13 by sl1pkn07