MarlinFirmware/Marlin #14929: [BUG] (short description)

mtowara, 2019-08-12:
### Description

I tried using the new SPI_ENDSTOPS feature on the x and y axis with TMC 2130s on the SKR 1.3 (LPC 1768 based), using Software SPI.

I encountered the following issues:
the spi calls within test_stall_status() in tmc_util.h do not return any (non zero) data for me.
The TMCStepper library makes a dummy SPI call first, maybe to flush out some buffer.
This code could be copied to here, I decided to just directly use the TMCStepper library:
```
if (this->TMC_SW_SPI != nullptr) {
    TMC2130_n::DRV_STATUS_t drv_status{0};
    drv_status.sr = this->DRV_STATUS();      
    sg_result = drv_status.stallGuard;
}
```

With this fix test_stall_status now returns true when the motor stalls.
( I think the spi communication problems also apply to M122 but this would be a seperate issue)

However the Endstops::update() routine in endstop.cpp seems to discard the information returned by test_stall_status(). I chased this down to the following:
```
  #if HAS_X_MIN
      #if ENABLED(X_DUAL_ENDSTOPS)
        UPDATE_ENDSTOP_BIT(X, MIN);
        #if HAS_X2_MIN
          UPDATE_ENDSTOP_BIT(X2, MIN);
        #else
          COPY_LIVE_STATE(X_MIN, X2_MIN);
        #endif
      #else
        UPDATE_ENDSTOP_BIT(X, MIN);
      #endif
  #endif
```
UPDATE_ENDSTOP_BIT will overwrite the live endstop reading with a read from the configured endstop pin and the endstop hit condition will never trigger.
This can be fixed by 

- `#if HAS_X_MIN && !X_SPI_SENSORLESS`

- or UPDATE_ENDSTOP_BIT such that it retains a set bit from the live endstops

- or sanity check could be altered to allow disabling USE_XMIN_PLUG, which it does not allow at the moment.

I would like to hear your opinion on whats the best fix here, once we've reached a verdict i would happily create a PR.

### Steps to Reproduce
enable SENSORLESS_HOMING and SPI_ENDSTOPS feauters with TMC2130

**Expected behavior:** Homing stops on motor stall

**Actual behavior:** Homing continues until printer is killed by homing distance safeguard

I would expect the first part of the problem to be potentially limited to LPC1768, the second part should be general.

McVillano, 2019-08-12:
I´m in the same situation i Think, I have defined SENSORLESS_HOMING and SPI_ENDSTOPS for a delta but the motors doesn´t stops until i cut the power. I seems it´s not working. Also Bigtreetech SKR 1.3 and Bigtreetech TMC 2130 V2

sl1pkn07, 2019-08-12:
is knowed

https://github.com/MarlinFirmware/Marlin/pull/14044

mtowara, 2019-08-12:
Thanks. That PR discussion is pretty long and github decides to hide some of it by default, so I did not fully read it before.
However, my specific issue is not addressed in there as far as I can see.

sl1pkn07, 2019-08-14:
>However, my specific issue is not addressed in there as far as I can see.

yep, you issue is more speciffic, but the base is the same, spi_endstops not working. in my case, never stop

greetings

rusnob, 2019-08-17:
Hello

Did you solve this problem?

I'm waiting for TMC2209 and the SKR 1.3

I want to make parking on the X and Y axes.

mtowara, 2019-08-17:
the 2209 are uart and not spi as I understand, so that feature won't help
you.

McVillano, 2019-08-17:
Mine finally was a pins problem, i own a delta printer that homes in ZMAX and the firmware was expecting the signal in ZMIN that is hardwired to DIAG1 in SKR 1.3

sl1pkn07, 2019-08-17:
> Mine finally was a pins problem, i own a delta printer that homes in ZMAX and the firmware was expecting the signal in ZMIN that is hardwired to DIAG1 in SKR 1.3

then you use wired sensorless instead of spi sensorless (spi sensorless have Zero wires rute to endstop pins)

McVillano, 2019-08-17:
When i say hardwired i mean inside board electronics its skr 1.3. But as all the cartesian printers home un zmin i had to change min for Max pins in board.h 

Just because its a delta printer

sl1pkn07, 2019-08-17:
yep. i know, but with spi sensorless not need that hard/wire

McVillano, 2019-08-17:
I'm pretty sure i'm using spi.

sl1pkn07, 2019-08-17:
try to retire the jumper

rusnob, 2019-08-17:
the question is not the topic but still
Which driver is better than tmc2209 or tmc2230?

AnHardt, 2019-08-17:
@rusnob 
> Which driver is better than tmc2209 or tmc2230?

That's a meaningless question. You have at firs to answer the question what is your definition of better.
Or in other words:

**It depends!**

sl1pkn07, 2019-09-24:
Hi. this error like expose @mtowara IDK, but SPI endstopless (withour hardwired/wired from diag1 pin) not work for me

greetings

mtowara, 2019-09-25:
@boelle 
I don't see any commits addressing this problem, so I guess not.
I'm running my own fix, as outlined in the initial post.

For my core xy I needed to also fix some other stuff for sensoless homing, I'll try to do a PR when I find some time.

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

sl1pkn07, 2019-10-12:
this not solved

sl1pkn07, 2019-10-12:
@mtowara can you share you own fix for this issue?

greetings

boelle, 2019-10-12:
> I'm running my own fix, as outlined in the initial post.

but i guess that is a bit "blured"

sl1pkn07, 2019-10-12:
if working, i can use it until get the proper fix

boelle, 2019-10-12:
and someone could make an PR so that it can get fixed faster :-D

mtowara, 2019-10-14:
@boelle your wish is my command ...

sl1pkn07, 2019-10-15:
https://github.com/MarlinFirmware/Marlin/pull/15560 not works for me

https://github.com/MarlinFirmware/Marlin/pull/15560#issuecomment-542393421

mtowara, 2019-10-17:
@sl1pkn07 sorry its not working for you.
I'll try to get to the bottom of this, but I only got ATMega and LCP1768 boards to test with.
Are you running with Hardware or Software SPI mode?

sl1pkn07, 2019-10-17:
Software, the hardware spi one is for the sd card (bundled in the board) and other for the LCD

boelle, 2019-10-22:
so the PR was merged, @mtowara i guess you are still working on this?

mtowara, 2019-10-22:
I got a spare SKR 1.3 lying around now, so hopefully I can use that to better diagnose things. But yes, the issue seems to be not fully fixed yet.

boelle, 2019-10-30:
@mtowara have you tested with the spare board?

mtowara, 2019-11-03:
I have, I still see the same behavior but am unable to recreate sl1pkn07's problem.
I updated the original PR which at least makes the changes for SPI_ENDSTOP less intrusive.

boelle, 2019-11-11:
@mtowara i assume this is still an issue?

closed 2019-11-21 by boelle

Allarabie, 2019-12-16:
This is millionth thread I've found on SPI Endstops not working using SKR1.3 with TMC2031, and noone has given any acceptable/working answer on how to fix this.

Has ANYONNE gotten Sensorless homing to work?

mtowara, 2019-12-16:
yes. However my fix for SPI_SENSORLESS seems not to work universally. Regular sensorless homing with diag pin connected to the x/y endstops works as well.

Allarabie, 2019-12-16:
Ok, finally found something that worked.!!!
**#define MINIMUM_STEPPER_PULSE 2** is Wrong!.
2 is NOT the correct pulse value for TMC2130s, 0 is.. 
Or comment this line out all together and use default value.
This is only thing that actually made sensorless homing trigger. Until I commented this out, nothing was making a difference and StallGuard was never triggering.

Also, separate issue (and FYI for Devs), is the homing code for Delta is also not correct for Sensorless homing.  See Thinkyhead's comment from Mar 17 2018. https://github.com/MarlinFirmware/Marlin/issues/10141
This is still incorrect in Marlin 2.0. However it is now in the delta.ccp file.

I know this isn't a chat room, but I hope this solution also helps others.

mtowara, 2019-12-17:
Good find!
For what its worth, sensorless homing on CoreXY is not ideal either.
Do you have a solution for Delta? Maybe one could coordinate a common PR.

sl1pkn07, 2019-12-17:
[image]

i have already set '#define MINIMUM_STEPPER_PULSE 0' 

and 

[image]

in the configuration_adv.h, and still not works for me :( 

greetings

Allarabie, 2019-12-17:
Sorry to hear that, it was very difficult to get working. Once I got Sensorless Homing working I tried to also use Sensorless Probing. 
I was only able to get either Sensorless homing OR Probing, but not both.
I finally gave up on sensorless homing, since I already had endstops switches on each tower.
And decided to just user Sensorless Probing.  
But I had to disable SPI_ENDSTOPS to get _anything_ Sensorless to work. So whatever I managed to do, it may/or may not be using SPI.

So Lets see in, config_adv:
Disable #define SOFTWARE_DRIVER_ENABLE
Disable #define SPI_ENDSTOPS (?)
Disable #define MINIMUM_STEPPER_PULSE
All the CHAIN_POS are set to -1 (default I think)

In the "pins" file I reversed the Endstop pins, so that Min is Max and Min is Max.
#if X_HOME_DIR == -1 || !X_STALL_SENSITIVITY
  #define X_MIN_PIN          P1_29   // X_MIN
  #define X_MAX_PIN          P1_28   // X_MAX
#else
  #define X_MIN_PIN          P1_29   // X_MAX
  #define X_MAX_PIN          P1_28   // X_MIN
#endif
This way it will use the correct endstop when probing. If not, it just crashes without triggering.
There is a note about this in the Pins file.  I also implemented the ThinkyHead fix I mentioned above to get Delta homing to work properly.

I've been screwing with this for 3 full days now, so Im sure I changed something else, that I forgot about. But hopefully this will get you going.
No clue if my setup is using SPI or not, but I have it (mostly) working, so Im not going to complain too much..

sl1pkn07, 2019-12-17:
if you have wired the diag1 pin in the driver to the board endstop header/s, then you not use spi_sensorless

the SPI sensorless should be work without wired diag1 driver pin (if use any BTT SKR "DIP" (interchangeable driver) variant, without the endstop red jumper)

the chain_pos is only if you use the TMC SPI daisy chain wire config (see https://github.com/MarlinFirmware/Marlin/pull/15081, also, not work for me)

Allarabie, 2019-12-17:
Everything I read says that I didnt have to wire anything on the BTT SKR1.3, I just needed to add the Red jumpers to the Diag Pins. So I did, and none of the SPI stuff worked. 
The TMC2130 v3 I got, already have the header soldered in to connect to the board. 
So I didnt need the extra wire to the endstop header, so Im told.. 

So, either Im still doing something wrong.
Or, the information about setting up SPI on this board is incorrect.
Or something is wrong in Marlin.

I guess I wasn't too much help. sorry..

sl1pkn07, 2019-12-17:
in the SKR 1.3, exist a jumper pin header dedicated for the sensorless feature (near the 12864 display headers). that pinheaders bridge the diag1 driver pin (two red headers, in vertical position, under the driver) to the endstops board conectors

this is the wire i talk. in the SKR boards the wire is not a external wire, is a internal track in the board. and open/close the circuit througth that jumpers

Allarabie, 2019-12-17:
I know there is no need for a physical wire, thats one of the reasons I got hit board.
The board came with those jumpers already installed. And all the info I found says they need to be installed.
"Once done, I can plug the Bl-Touch in Z-Min, unplug the two other endstop and add the XST, YST, ZST, E0S jumpers "
But you're saying to Connect SPI, I have to disconnect the ST jumpers?
Seems weird, but Im willing to try it.

Allarabie, 2019-12-17:
Yunno what would be really cool?
If someone (anyone) who either wrote this code, or knows how to set it up properly, could chime in and tell us all how its supposed to be implemented.!?  Because so far, all Im finding is conflicting instructions, and none of them seem to deliver a usable result. 
It would save us all allllllllllot of time and frustration.
Thanks in advance.

mtowara, 2019-12-17:
@Allarabie from your previous posts I see that you generally know how to do it, I think the problem is that the current implementation is just not suitable for deltas or core xy. I'm leaving my more general answer below as well, as there seems to be some misconception what the SPI_ENDSTOP feature is actually doing as opposed to connecting diag to the endstop pins.

On SKR, if the jumpers are in place the SPI_ENDSTOP feature is not needed, as there is a physical connection from the diag output of the drivers to the endstop inputs of the SKR board. However still a working SPI connection is required to setup the correct driver modes. So make sure that TMC_USE_SW_SPI is enabled and try to debug the drivers with M122.

Allarabie, 2019-12-17:
Thanks @mtowara.  
There is, in fact, a lot of conflicting information out there on how to get Sensorless stuff working.  I think in most significant confusion I find, is the information out there is not clearly separated between "Sensorless homing/probing" and using SPI for Sensorless features (or maybe Im the one thats confused and the 2 arent even related). 
Then, obviously, there are countless boards and driver combinations, it makes it very difficult to find info on the exact setup you're working with.
An other thing that confuses me is the #define SOFTWARE_DRIVER_ENABLE feature.
Its description says to Enable this to use SPI capable drivers.  But all this does is disable the drivers all together (probably because I have some other setting incorrect).  

Anyway, I have TMC_USE_SW_SPI enabled, and the jumpers in place.
And I am able to get debug info from M122. But nothing seems out of place.  Not that I really know what Im looking for, other than the obvious connection problems.

So, If im understanding this correctly, I need:
TMC_USE_SW_SPI = enabled
SOFTWARE_DRIVER_ENABLE = disabled
SPI_ENDSTOPS = enabled
And remove the ST jumpers
?

I also found one confusing post that stated that the INT/+5V jumper needs to be on the 5V side for SPI/UART to work on this board.. The reason, they stated, was that the SPI/UART circuitry needs 5V to function.  

Anyway, Im just rambling now.. 
Ill keep messing with it and see what happens.

sl1pkn07, 2019-12-17:
is simply:

SPI_ENDSTOPS enable? = enstops jumpers out
SPI_ENDSTOPS  disable? = endstop jumpers in

SPI_ENDSTOPS  only work in SPI drivers, like the TMC2130 and TMC5160 (and other SPI drivers)

the only avantage of SPI_ENDSTOPS is if you use a board without "hardwired" the diag1 pin to the endstop pins

in the some SKR family boards (not all) not need this feature (like ponts @mtowara)

Allarabie, 2019-12-17:
Yep, I know all this, and have read this same thing in posts over and over again. 
And I have tried every combination of those settings, I can think of..  
And none of it works, the printer just crashes into the towers, or the bed.
So sadly its not as simple as just changing those few settings.

sl1pkn07, 2019-12-17:
What driver is? Have the diag1 pin welded In the driver?

Allarabie, 2019-12-17:
TMC2130
Yes the pins were already soldered to the driver.
This is the exact setup.
https://www.amazon.com/gp/product/B07WDWKLQK/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

nick777745, 2019-12-29:
Have you been able to get the tmc2130 working with sensorless homing?

Allarabie, 2019-12-30:
TL;DR:  No!

Not exactly. I was able to get X an Y to sensorless home without issue, but not Z (delta).
I have been more focussed on sensorless probing, which I also got to "sorta" work. But the values it gets for the endstop offsets are completely wrong.  The Z endstop offset is off by almost 2mm and the effector ends up crashing into the bed when moving towards the Z tower.  
Lets say I send the printer G1 Z2 Y-80 X0, the effector will move to the front of the front of the print area but be almost 5mm away from the bed, then I send G1 Z2 Y80 X0, the effector will move to the back of the bed and slowly gets closer and closer to the bed till is crashes into it and starts losing steps.  Also, at no point does StallGuard kick in and stop the printer, it just keeps scraping along the bed destroying my print surface and and losing steps till I kill the printer (assuming my understanding of StallGuard is correct).
But wait, there's more.
I have the delta homing with regular endstop switches. X and Y home smoothly and quietly, but Z moves violently and smashes into the endstop.
All and All, I would say that Delta support really needs to be reviewed, not just the Sensorless portions but all the homing and probing code. 
I also tried just using the force sensitive resistor that came with the printer (similar to this: https://upload.wikimedia.org/wikipedia/commons/thumb/c/c9/Force_Sensing_Resistor_cropped.jpg/1200px-Force_Sensing_Resistor_cropped.jpg), and I get the same incorrect offsets afterwards, this is why I suspect the Delta homing/probing code needs more review.