MarlinFirmware/Marlin #14930: Z is not obeying DEFAULT_MAX_FEEDRATE

DarkAlchy, 2019-08-12:
Found the solution to my problem that I did mention before but now I have confirmed it.

When I used Marlin 1.1.7, or Repetier, this line always capped my feedrate (go back to beta circa 2012 of Marlin) #define DEFAULT_MAX_FEEDRATE          { 200, 200, 6, 50 }

That is my feedrates but in Slic3r and S3D I had Z set as 15mm/s.  No issues until 2.0 due to it ignoring that cap.  Is this the new intended way or is something broken?

thisiskeithb, 2019-08-12:
Currently, it's not a bug. See PR https://github.com/MarlinFirmware/Marlin/pull/13929 for more info.

DarkAlchy, 2019-08-12:
Ahhh, so you have changed the behaviour then from what I am reading so working as intended for 2.0 but not if on the older branches?

DarkAlchy, 2019-08-12:
Closing this one out since there is movement on issue #13929

closed 2019-08-12 by DarkAlchy