MarlinFirmware/Marlin #14932: [BUG] Commit f8aa52346 doesn't fit 1284p with BLTouch enabled

trouch, 2019-08-12:
### Description

I've noticed that I can't build a BLTouch enabled firmware for Ender-3 stock board anymore using latest code base. Trying to build a translated firmware is also funny, even without bltouch enabled !

I reduced the features as much as possible without any luck, config attached.

So I slowly reverted back until I found the commit that brought more than 1.5k of extra binary code:

`Mixer patch, etc.	842466f4d	Scott Lahteine <github@thinkyhead.com>	30 Jul 2019 at 05:20`
=> Uses 128906 bytes, fits Ender-3 stock board

`Configurable Z move distance (#14593)	f8aa52346	MarcelMo <mommsen@gmx.de>	30 Jul 2019 at 09:16`
=> Uses 130628 bytes, goes over bootloader space

### Steps to Reproduce
1. Checkout f8aa52346 or following commit.
2. Uses attached configuration
3. Build using either melzi_optiboot or sanguino_1284p PIO env

**Expected behavior:** Build should fit

**Actual behavior:** Fails

#### Additional Information

PR is following.

reloxx13, 2019-08-12:
heya,

- disable bootscreen
- use JAPANESE charset
- disable sdsupport
- disable autotemp
- disable lcd info screen
- disable babysteps
....

seriously, did u even watch one guide about bltouch and ender 3 configuration ?

even before this commit it was hard to enable bltouch and sdsupport, u had to disable alot. but u did not even disable those "basics"...
u yeah, u disabled ARC_SUPPORT, thats it. it will never fit like this.

pls disable everything and report back or buy a better board. there is no other solution.

reloxx13, 2019-08-12:
PS:

> Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.
> 
> Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

InsanityAutomation, 2019-08-12:
@reloxx13 maybe he gave too much information. Let me revise this : Commit ## grows binary size by x.xkb, here is a pull request to make new functionality optional so that a finely tuned build still fits where it did before if I dont want the new feature added.

reloxx13, 2019-08-12:
@InsanityAutomation but the config is not fine tuned, cause bltouch and sdsupport did never fit without disableing alot of stuff for small chips. and without sdsupport there is(was) alot of space left.

there are special guides out there to enable sd support and bltouch (for ender3), which shows to disable those listed above (maybe not all, but he did not disable one of those).

if the size grew now, u may be forced to eg. disable babysteps now, which u could enable before this commit.

start with disabling all of those listed and start enabling features u like and check the size everytime u enable something.

InsanityAutomation, 2019-08-12:
@reloxx13 Im well aware of that, ive been maintaining a branch with BLTouch, SD Support, and Mesh leveling / bilinear for quite some time. Eric has seen those and may have even cherry picked some of it out for his own setup testing here. Goal is to turn off as little as possible. Last I checked my config compiled at 95% progmem (96.5% hits bootloader!).

In any case, all that is just a distraction from new functionality added not in the normal convention of a config option to turn it off increasing global size.

trouch, 2019-08-12:
@reloxx13 Come on... don't take it that bad. Please let met add few more details.

In order to debug the issue with my own fine tuned config, I started from the sample Ender-3 config, enabled BLTouch, and started to strip down things. At some point I realized something in the tree expanded the firmware too much.

I've spent lot of time to find the exact commit and feature which increased the footprint.

I obviously know we can't fit everything on a small 128kB memory. Thought I pointed out an issue that lot of Ender-3 user may encounter at some point, and even provided a PR for it.

sl1pkn07, 2019-08-12:
i have the same problem in my 1.1.3 creality board (mine without bltouch). before, (1088846caef6f660d5af65e1725a6a6deeef16fd) with my config, all fit in the board, 

now i have rebased the code with the new commits (13d8dc0b79d650312a1a24c7e59421d3121a8a1b) i'm unable to fit the firmware because exceed the board size

`Error: The program size (130572 bytes) is greater than maximum allowed (130048 bytes)`

yes is a litle, but before fits without problem, now not

my changes (before rebase (not pushed) https://github.com/sl1pkn07/Marlin/commit/8641aca1963d1987bf9c6ed0338e31baa4d899ca)

greetings

trouch, 2019-08-13:
@sl1pkn07 give a try with `#define SHORT_MANUAL_Z_MOVE 0`

closed 2019-08-13 by trouch